ARMmbed/yotta #134: Suggested name when using 'yt init' is always lowercase

thegecko, 2015-01-05:
It seems having a yotta module with a different name in the module.json to the folder name is incorrect:

Exception: Component pocketsvg found in incorrectly named directory PocketSVG (/Users/robmor01/Projects/envoy-ios/yotta_modules/PocketSVG)

Therefore the suggested name when running 'yt init' should match the folder name you are running in, but instead it suggests a lowercase version of the same:

PocketSVG $ yt init
=>
Enter the module name: \<pocketsvg\> // should be PocketSVG

thegecko, 2015-01-05:
Modules being lowercase could cause a problem. The reason I saw this was due to me checking out a fork of a 3rd party library, already mixed case. I can't think of a nice way to make such a project folder lowercase.

autopulated, 2015-01-06:
The top-level directory name shouldn't matter, but that doesn't mean the module name convention should change. That check just needs to be weaker.

fwiw you can also do `git clone <source-url> somename` to check out a repo into whatever named directory you like.

thegecko, 2015-01-07:
Yeah, a weaker check would work.

autopulated, 2015-02-03:
I can't reproduce this problem.
Do you have a minimal set of instructions to reproduce that doesn't involve renaming things inside yotta_modules?

closed 2015-02-03 by autopulated