ARMmbed/yotta #135: Yotta shouldn't build all xcode targets by default, perhaps not tests

thegecko, 2015-01-07:
*In the context below, 'target' is an xcode target, not a yotta architecture*

As it stands, building ios apps requires signing (unless building for a simulator).
If an Xcode build is undertaken using yotta (`yt -G Xcode`), the signing is automatic as long as every target has a valid BundleIdentifier.

Yotta seems to build for all targets by default, including all test targets in any dependency modules, many of which don't have Info.plist files to house a valid BundleIdentifier (eris, yttimer, etc.), so errors of this form are generated:

`Code Sign error: Automatic provisioning profile selection unavailable: A bundle identifier is required for automatic provisioning profile selection. Either enter a bundle identifier in the Info.plist...`

Opening the Xcode project and building from there works, though.

Options:

-Make yotta not build test targets (preferable)
-Allow yotta to only build a specified target (perhaps a `-target <target-name>` flag)
-Change yotta just to generate Xcode project files without running a build requiring developers to build in Xcode (eww)
-Ensure any targets have Info.plist files and suitable BundleIdentifiers (unnecessary work IMO)

autopulated, 2015-01-08:
Is there a way to do this using `yotta build -- <build-tool-options>`?

thegecko, 2015-01-12:
Yes:

```bash
yotta build -G Xcode -- --target <target-name>
```

closed 2015-01-12 by thegecko