ARMmbed/yotta #161: Access to mbed ARMmbed/mbed-sdk

mikes-spindance, 2015-02-25:
Hello,

Is there a yotta email list?

At SpinDance, http://www.spindance.com/,  we are working to be early adopters of the new yotta build system.

Please email me at mikes@spindance.com 

I'm getting an access issue when trying to cross-compile.

````
tomato:helloyotta mikes$ yotta build
info: get versions for frdm-k64f-gcc
info: get versions for mbed
info: failed to fetch Github object, re-trying with authentication...

You need to log in with Github.
Press enter to continue.
Your browser will open to complete login.

waiting for response...
If you are unable to use a browser on this machine, please copy and paste this URL into a browser:
http://yottabuild.org/#login/c3NoLXJzYQAAAAMBAAEAAAEBALuG3Iv0AM4hBbL/Oxp7W/dNzDWXlRJ9pVyVas/H32NwFQv8XaMFnQHeWyqyCP3XhE/JVkM1wDGpgAAQBow785Kc4vBuS08g%2BU7ChhtpSOsubm8MZBekFYlUGN4r4to8n76DMYhPZXX7YxDjgN7thidzumsizqLC6HjFuk/5hWNFwtNzYtey5NVu9qgqF3FU6xe2hggmo9Vz4ls5ODpmczeDEEV1kb54iV/w3UgGipxwM5NiSxt6ECvpNMKw58ipMA5pE35KJ6W8/TG47rLRx24Ppyxqq6%2BSKY2/Z39JlzMe2LPlT1CgCyFp8Iau5PIUfC/SMcW4%2BSUTkDkt/vsH0tE%3D?provider=github
......
error: Failed to satisfy dependencies of /Users/mikes/Documents/spindance/davisville/embedded/testYotta/helloyotta:
error: could not locate github component "ARMmbed/mbed-sdk", either the name is misspelt, you do not have access to it, or it does not exist
error: mbed not available
error: Missing dependencies prevent build. Use `yotta ls` to list them.
tomato:helloyotta mikes$ yotta ls
helloyotta 0.0.0
┠─ simplelog 0.0.1
┗─ mbed ARMmbed/mbed-sdk missing
````

autopulated, 2015-02-25:
Hi Mike!

The version of mbed that works with yotta is currently in private alpha testing. I'll put someone in touch with you who'll be able to talk to you about getting access to that that.

jacobrosenthal, 2015-02-25:
Can I be considered as well?

autopulated, 2015-02-25:
@jacobrosenthal We're currently only set up to provide early access to companies (sorry!), but we do expect to have a public beta in the future, too.

sg-, 2015-02-25:
Early access is for mbed partners (who are companies :) Early developer access will be available later. To become a partner you can send a request to partnership@mbed.org

BlackstoneEngineering, 2015-07-31:
We now have more stuff accessible on the public registry. For reference mbed 2.0 stuff is now called mbed-classic. Go give it a shot and try building stuff!

closed 2015-08-06 by autopulated