ARMmbed/yotta #162: windows - required to create folders in Program files ?

0xc0170, 2015-03-02:
The #62 mentioned that it links under Program files. I am receiving the following error:

```
Traceback (most recent call last):                                                                             
  File "C:\Python27\Scripts\yotta-script.py", line 9, in <module>                                              
    load_entry_point('yotta==0.0.43', 'console_scripts', 'yotta')()                                            
  File "C:\Python27\lib\site-packages\yotta-0.0.43-py2.7.egg\yotta\main.py", line 135, in main                 
    status = args.command(args, following_args)                                                                
  File "C:\Python27\lib\site-packages\yotta-0.0.43-py2.7.egg\yotta\link.py", line 58, in execCommand           
    fsutils.mkDirP(folders.globalInstallDirectory())                                                           
  File "C:\Python27\lib\site-packages\yotta-0.0.43-py2.7.egg\yotta\folders.py", line 24, in globalInstallDirect
ory                                                                                                            
    fsutils.mkDirP(dirname)                                                                                    
  File "C:\Python27\lib\site-packages\yotta-0.0.43-py2.7.egg\yotta\lib\fsutils.py", line 14, in mkDirP         
    os.makedirs(path)                                                                                          
  File "C:\Python27\lib\os.py", line 150, in makedirs                                                          
    makedirs(head, mode)                                                                                       
  File "C:\Python27\lib\os.py", line 157, in makedirs                                                          
    mkdir(name, mode)                                                                                          
WindowsError: [Error 5] Access is denied: 'C:\\Program Files\\yotta'                                           
```

I solved it running the cmd as administrator (=this report might help somebody who is facing this error). Wouldn't be better to replace it by placing yotta directory outside of Program Files as a work around, might not require "windows sudo"?

autopulated, 2015-03-02:
OK, that doesn't sounds like the best place. Where should such things go on windows? (for reference, the global link directory on linux is `/usr/local`, but it should _probably_ also be changed to use a user-specific one, perhaps in `~/.yotta/install`.

bogdanm, 2015-03-03:
This isn't trivial. The most obvious solution would be indeed ```$HOME/.yotta```, since ```HOME``` also exists in Windows. However, in Windows, ```HOME``` can point to a network share and I don't think it's possible to create a link on a network share (is this also true in Linux if ```HOME``` is a NFS mount or similar?). We can guess more directiories in Windows, but as a generic solution I'd add a ```YOTTA_HOME``` environment variable than can be set by the user, with the default value ```$HOME/.yotta```.

autopulated, 2015-03-03:
for reference, the way that npm's paths are specified is documented [here](https://docs.npmjs.com/files/folders).

autopulated, 2015-03-19:
This was fixed in #176

closed 2015-03-19 by autopulated