ARMmbed/yotta #163: yotta ls - the output in command lines (OS: windows)

0xc0170, 2015-03-02:
This is the output for git bash, cmd or mingw. 

```
ÔöáÔöÇ xx 0.0.5
Ôöâ ÔöùÔöÇ FOLDER1 0.0.2 yotta_modules\FOLDER1
Ôöâ   ÔöùÔöÇ FOLDER2 0.0.4 yotta_modules\FOLDER2
Ôöâ     ÔöùÔöÇ FOLDER3 0.0.1 yotta_modules\FOLDER3
Ôöâ       ÔöáÔöÇ FOLDER4 0.0.2 yotta_modules\FOLDER4
ÔöùÔöÇ MODULE1 0.0.1
  ÔöùÔöÇ MODULE2 0.0.1 yotta_modules\MODULE2
    ÔöùÔöÇ MODULE3 0.0.2 yotta_modules\MODULE3
```

It's readable though, but probably not desirable :-)

autopulated, 2015-03-02:
Unicode in terminals :sad:

0xc0170, 2015-03-02:
Small addition, I haven't got any problems in ubuntu. So would like to know how to fix this for windows based terminals

closed 2015-03-02 by autopulated

0xc0170, 2015-03-02:
Thanks for immediate fix! Looking better now:

```
mbed-example-as
|_ module1
| |_ module2
| | \_ module3
| |   \_ module4
```

autopulated, 2015-03-02:
no problem :+1:

commit db09a340 (2015-03-02): Merge pull request #164 from autopulated/windows-encoding-fixes

don't use unicode for tree printing on windows; fixes #163
commit eca3b874 (2015-03-02): don't use unicode for tree printing on windows; fixes #163