ARMmbed/yotta #167: yotta clean fails on windows 7

simonqhughes, 2015-03-04:
Hi

environment: windows 7, cygwin
problem: yotta script faults when tries to do clean

here is the traceback:

simhug01@E103032 /cygdrive/c/develop/public/jobs/yr2015/2209/work/20150304/proj/mbed-example-asynch-spi
$ yotta clean
Traceback (most recent call last):
  File "C:\mbed_tools\Python27\Scripts\yotta-script.py", line 9, in <module>
    load_entry_point('yotta==0.0.43', 'console_scripts', 'yotta')()
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\main.py", line 135, in main
    status = args.command(args, following_args)
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\clean.py", line 23, in execCommand
    fsutils.rmRf(os.path.join(c.path, 'build'))
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\lib\fsutils.py", line 28, in rmRf
    shutil.rmtree(path)
  File "C:\mbed_tools\Python27\lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\mbed_tools\Python27\lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\mbed_tools\Python27\lib\shutil.py", line 256, in rmtree
    onerror(os.rmdir, path, sys.exc_info())
  File "C:\mbed_tools\Python27\lib\shutil.py", line 254, in rmtree
    os.rmdir(path)
WindowsError: [Error 145] The directory is not empty: 'C:\\develop\\public\\jobs\\yr2015\\2209\\work\\20150304\\proj\\mbed-example-asynch-spi\\build\\frdm-k64f-gcc\\CMakeFiles'

simhug01@E103032 /cygdrive/c/develop/public/jobs/yr2015/2209/work/20150304/proj/mbed-example-asynch-spi

Simon

autopulated, 2015-03-05:
Is it possible that you had a file in the build directory open/locked by an editor?

autopulated, 2015-03-05:
(closing because I can't reproduce, I suspect a file in the build directory was locked due to an open explorer window, or editor)

closed 2015-03-05 by autopulated