ARMmbed/yotta #184: yotta build fails on windows 7, github problem?

simonqhughes, 2015-03-20:
simhug01@E103032 /c/develop/public/jobs/yr2015/2209/work/20150320/mbed-sdk (master)
$ yt target
frdm-k64f-gcc,*
←[0m
simhug01@E103032 /c/develop/public/jobs/yr2015/2209/work/20150320/mbed-sdk (master)
$ yt build
←[22m←[32minfo←[39m: ←[22m←[39mget versions for mbed-hal←[0m
Traceback (most recent call last):
  File "c:\mbed_tools\Python27\Scripts\yt-script.py", line 9, in <module>
    load_entry_point('yotta==0.1.0', 'console_scripts', 'yt')()
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\main.py", line 150, in main
    status = args.command(args, following_args)
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\build.py", line 50, in execCommand
    install.execCommand(args, [])
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\install.py", line 50, in execCommand
    installDeps(args, c)
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\install.py", line 88, in installDeps
    available_components = [(current_component.getName(), current_component)]
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\lib\component.py", line 447, in satisfyDependenciesRecursive
    provider = provider
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\lib\component.py", line 282, in __getDependenciesRecursiveWithPr
vider
    provider = provider
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\lib\component.py", line 190, in __getDependenciesWithProvider
    satisfyDep, specs
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\lib\component.py", line 179, in satisfyDep
    update_installed
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\lib\component.py", line 436, in provider
    r = access.satisfyVersionByInstalling(name, version_req, self.modulesPath())
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\lib\access.py", line 283, in satisfyVersionByInstalling
    v = latestSuitableVersion(name, version_required)
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\lib\access.py", line 120, in latestSuitableVersion
    vers = remote_component.availableVersions()
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\lib\github_access.py", line 254, in availableVersions
    for t in self._getTags():
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\lib\github_access.py", line 244, in _getTags
    self.tags = _getTags(self.repo).items()
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\lib\github_access.py", line 57, in wrapped
    return fn(*args, **kwargs)
  File "C:\mbed_tools\Python27\lib\site-packages\yotta\lib\github_access.py", line 83, in _getTags
    repo = g.get_repo(repo)
  File "C:\mbed_tools\Python27\lib\site-packages\github\MainClass.py", line 200, in get_repo
    "%s%s" % (url_base, full_name_or_id)
  File "C:\mbed_tools\Python27\lib\site-packages\github\Requester.py", line 169, in requestJsonAndCheck
    return self.__check(*self.requestJson(verb, url, parameters, headers, input, cnx))
  File "C:\mbed_tools\Python27\lib\site-packages\github\Requester.py", line 177, in __check
    raise self.__createException(status, responseHeaders, output)
github.GithubException.GithubException: 403 {u'documentation_url': u'https://developer.github.com/v3/#rate-limiting', u
message': u"API rate limit exceeded for 217.140.96.140. (But here's the good news: Authenticated requests get a higher
ate limit. Check out the documentation for more details.)"}
←[0m
simhug01@E103032 /c/develop/public/jobs/yr2015/2209/work/20150320/mbed-sdk (master)
$

bogdanm, 2015-03-20:
Try doing 'yt login'

autopulated, 2015-03-20:
Yes, this is caused because you're not authenticated, and have exceeded the request rate limit.

closed 2015-03-20 by autopulated