ARMmbed/yotta #186: Machine-readable yotta ls output

bremoran, 2015-03-20:
Some common operations would be easier if `yotta ls` produced machine-readable output.  In particular, a way of removing the tree diagram and colour would help.

For example an output like the following would provide easy parsing
```
$ yotta ls
module#version
dependency1#version
linked-dependency -> link-target#version
```

autopulated, 2015-03-20:
`--plain`?

bremoran, 2015-03-20:
--porcelain?
See: [git-status](http://git-scm.com/docs/git-status) --porcelain

hugovincent, 2015-03-20:
I literally shuddered just now trying `--porcelain?` as output reminded me of SVN...

danros, 2015-03-20:
This is something we will need also, for all yotta commands really.

-- IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

ARM Limited, Registered office 110 Fulbourn Road, Cambridge CB1 9NJ, Registered in England & Wales, Company No: 2557590
ARM Holdings plc, Registered office 110 Fulbourn Road, Cambridge CB1 9NJ, Registered in England & Wales, Company No: 2548782

autopulated, 2015-03-20:
So, some commands (like build) stream the output from subprocess as they run, and add their own output before/after that.

I'd like to add a `--json` type flag, which processes command output into a very robust machine readable form, but unfortunately that is not easily compatible with streaming output.

So here's what I propose.

`yotta` accepts `--plain` as a top level argument (before any subcommands). This will prevent coloured output in all commands, and modify the output of commands like `yotta list` to remove other fancy formatting.

If that proves insufficient, we can add additional `--json` output flags for specific commands where it makes sense.

In the future --plain might be the default if yotta detects stdout is a pipe, too.

bremoran, 2015-03-20:
That sounds good to me.

autopulated, 2015-04-27:
See #226

autopulated, 2016-03-22:
This was fixed since yotta 0.13.0, when the `yt ls --json` format was stabilised.

closed 2016-03-22 by autopulated