ARMmbed/yotta #193: Specify Python Version in Read.me

BlackstoneEngineering, 2015-03-23:
In in the "Installation" Section of the readme it needs to be specified and linked which version of python is being used.

## Currently it looks like this
### Installation
Detailed instructions can be found on the documentation site, in summary, to use yotta you need:
*    yotta itself (yotta is written in python, and installed using pip)

## should look like this

### Installation
Detailed instructions can be found on the documentation site, in summary, to use yotta you need:
*    yotta itself (yotta is written in python2.7, and installed using pip)

autopulated, 2015-03-23:
yotta should work with any recent enough version of python (2 or 3), but I'll add a recommendation for python 2 as the best tested version.

BlackstoneEngineering, 2015-03-23:
We should add a warning to windows users to go an read the documentation, 'cause for them the instructions on the read.me file are no where near enough (PyCrypto, ninja, the order of install, everything)

commit e189ac3a (2015-03-23): add recommended python version to readme; fixes #193
closed 2015-03-23 by autopulated