ARMmbed/yotta #195: Update references to Python2.9 (Windows problem)

BlackstoneEngineering, 2015-03-23:
All the documentation (where linked) points to python 2.8, it needs to be updated to point to 2.9, 

As of 2.9 pip is included in python, this makes the entire process that much easier. 
In 2.8 pip must be manually installed, manually added to path, not trivial, not easy.

autopulated, 2015-03-23:
I believe I've correctly updated all references to python 2.7.9. If there are specific references I've missed, let me know where they are!

BlackstoneEngineering, 2015-03-23:
Looks like you've caught all of them, only found one issue left: 

## https://github.com/armmbed/yotta
* If permission is denied, you may need to run sudo pip install yotta, and you may need to first install Python 2.7 and pip, if you do not already have them. (Change this to 2.7.9, the link is correct, the text is not)

It might be worth re-wording this entire section to remove the instructions  on installing pip, since its not longer necessary, pip is installed by default as part of 2.7.9, this should greatly simplify the instructions.

Also may be worth bringing the windows install script up.

autopulated, 2015-03-26:
Being explicit about 2.7.9 caused some issues on linux (where the system package manager provided an earlier version), so I've updated the documents to try to still be clear on windows, but suggest running with what you've got on other platforms.

closed 2015-03-26 by autopulated

BlackstoneEngineering, 2015-03-26:
Windows needs 2.7.9 because it included pip, everyone else with a proper package manager wont have an issue getting pip, or should already have it, but on windows... jeasus its bloody