ARMmbed/yotta #196: Add windows install script

BlackstoneEngineering, 2015-03-23:
There is a windows install script here: https://github.com/BlackstoneEngineering/yotta/tree/get_yotta_win_script 

It should be brought in as part of the release of yotta and periodically updated to make installing yotta on windows easier. Or at the very least provide it as an option. It simply downloads the dependencies and runs the installers in order. It assumes Python and Pip are installed on the system, this is a safe assumption as of Python 2.9 where pip and python are bundled together at install time.

autopulated, 2015-03-24:
There's an open pull request for this. Is that up to date?

BlackstoneEngineering, 2015-03-24:
There is a pull request, here https://github.com/ARMmbed/yotta/pull/132 , 

It is not up to date, it was put together for the alpha1 release and needs updated links (ie remove pycrypto, update cmake link) nothing major. The script was meant as a git_pip.py style yotta install script. There are hooking for adding commands to auto-install yotta for mac and linux as well. In theory this script could become the one stop shop for grabbing yotta.

-Austin

BlackstoneEngineering, 2015-03-24:
To clarify, I think this script should be added for Alpha2 testing in prep for actual release as part of the Beta, because it seems to be the only viable option for installing on windows, the current mess is not acceptable for a release to the general public. It is simply too complex.

autopulated, 2015-03-26:
I've merged the script. One question:

 * if the python version is 2.7.9, is it still necessary to open IE to download certificates? (Python 2.7.9 included updates to SSL certificate verification that might have fixed that)

closed 2015-03-27 by autopulated

BlackstoneEngineering, 2015-03-27:
Unclear, didnt test that case. It was an issue with the website and windows itself so ... maybe? I'll retest in the next OoB