ARMmbed/yotta #202: Yotta debug does not start on Freescale K64F in Alpha2

SeppoTakalo, 2015-03-26:
I'm trying to start the yotta debugger for Alpha2 example mbed-coap-example.

~~~
$ yt debug source/mbed-coap-example
warning: mbed-coap-example has invalid module.json:
warning:   repository.url value u'' is not valid under any of the given schemas
Traceback (most recent call last):
  File "c:\Python27\Scripts\valinor-script.py", line 9, in <module>
    load_entry_point('valinor==0.0.1', 'console_scripts', 'valinor')()
  File "C:\Python27\lib\site-packages\valinor\main.py", line 74, in main
    ide_tool = ide_detection.select(ide_detection.available(), args.target)
  File "C:\Python27\lib\site-packages\valinor\ide_detection.py", line 123, in select
    possible_ides = [x for x in available_ides if tool.target_supported(target, x)]
  File "C:\Python27\lib\site-packages\project_generator\tool.py", line 62, in target_supported
    return exporter.supports_target(target)
  File "C:\Python27\lib\site-packages\project_generator\exporters\uvision.py", line 173, in supports_target
    return target in self.definitions.definitions.mcu_def
AttributeError: uVisionDefinitions instance has no attribute 'definitions'
error: debug process executed with status 1
~~~

0xc0170, 2015-03-26:
This was bug in the project generator, fixed in the v0.4 https://github.com/0xc0170/project_generator/commit/657b6578d3e2c12a5b406716ff2acbf4bbe7c6ee

autopulated, 2015-03-26:
This should be fixed if you run `pip install -U yotta`

what are the outputs of:
```
valinor --version
project_generator --version
```

SeppoTakalo, 2015-03-26:
septak01@E104494 ~/mbedOS-alpha2/examples/mbed-coap-example
$ valinor --version
0.0.1

septak01@E104494 ~/mbedOS-alpha2/examples/mbed-coap-example
$ project_generator --version
0.3.1

SeppoTakalo, 2015-03-26:
OK, now I updated Yotta.

~~~
$ yotta debug source/mbed-coap-example
warning: mbed-coap-example has invalid module.json:
warning:   repository.url value u'' is not valid under any of the given schemas
Invalid parameter(s)
RESET { SESSION }
Traceback (most recent call last):
  File "c:\Python27\Scripts\yotta-script.py", line 9, in <module>
    load_entry_point('yotta==0.2.2', 'console_scripts', 'yotta')()
  File "C:\Python27\lib\site-packages\yotta\main.py", line 150, in main
    status = args.command(args, following_args)
  File "C:\Python27\lib\site-packages\yotta\debug.py", line 40, in execCommand
    for error in target.debug(builddir, args.program):
  File "C:\Python27\lib\site-packages\yotta\lib\target.py", line 206, in debug
    preexec_fn = _newPGroup
  File "C:\Python27\lib\subprocess.py", line 663, in __init__
    raise ValueError("preexec_fn is not supported on Windows "
ValueError: preexec_fn is not supported on Windows platforms
~~~

~~~
$ valinor --version
0.0.3
$ project_generator --version
0.4.1
~~~

autopulated, 2015-03-26:
ah ok, so updating yotta fixed the original issue, good.

autopulated, 2015-03-26:
Your new error is caused because the target description specifies a debug server, which is not supported on windows. `valinor` is designed to solve this problem, so this is a bug in the target description you're using (which may be out of date)

closed 2015-03-26 by autopulated