ARMmbed/yotta #209: Smarter windows intstall script

sg-, 2015-03-31:
See comment thread here. https://github.com/ARMmbed/yotta/pull/132#issuecomment-88221954

BlackstoneEngineering, 2015-03-31:
## Todo: 
* silent install of dependencies
* check for existence of dependencies before downloading
* ...?

BlackstoneEngineering, 2015-07-31:
@sg- @autopulated  we should probably schedule this for review soon for the public beta?

autopulated, 2015-08-03:
PRs welcome :)

My view on this is that the current instructions are sufficient, even though there could be fewer steps. I think that clearly explaining to people what the dependencies are and where they are installed is something that most people appreciate even if they have to manually perform a number of steps.

So yea: easier is always good, but what we have is OK.

sg-, 2015-08-27:
https://github.com/ARMmbed/yotta_windows_installer Thank you @BlackstoneEngineering This is a great 0.01 :)

BlackstoneEngineering, 2015-08-28:
Correction: heres an update https://github.com/ARMmbed/yotta_windows_installer/releases/tag/v0.1.0

autopulated, 2015-10-01:
resolved by #431 :tada: :balloon:

closed 2015-10-01 by autopulated