ARMmbed/yotta #210: windows yotta-var.bat

sg-, 2015-04-01:
Yotta should have a config file for paths to dependencies. This can be set by an install script or modified by a developer. Developers may have many versions of a tool installed on a computer and many in the path. This will enable a more controlled build process. When invoked yotta can have its own tmp path for a given session.

autopulated, 2015-04-01:
Are you suggesting we should ship a yotta-env.bat script that sets up the environment, or that yotta should set up the PATH for its subprocesses based on variables in the `~/.yotta/config` file?

I don't think putting PATH stuff in the yotta config file makes sense – I don't see how that's useful to a user, and it wouldn't make the multi-environment setup you describe any easier.

We could ship a yotta-env.bat script which sets up the command prompt for you to use yotta, but it wouldn't know the paths of the things it needs to set, *and* because future yotta targets can (and will) need new compilers and other tools not needed by the current set of targets, it could not possibly set up a working environment for every target.

So my conclusion is that this doesn't belong in yotta itself. Does that make sense?

Maybe each yotta target should have clear, separate, installation instructions for the tools it needs (at the moment they are sort-of merged into the yotta installation instructions for our supported targets), and those instructions could installers and env.bat scripts?

sg-, 2015-04-01:
>>Are you suggesting we should ship a yotta-env.bat script that sets up the environment
Yep - sets up the environment just as each yotta call is made. If yotta is in the path, compilers and dependencies dont need to be. They should be a config in yotta that is filled out by the installer based on location of installations or best guessed defaults (editable by developer). If you have a look at many tools that does what yotta does (build software) this is common practice. The registry and session modified PATH is your friend in controlling the build environment.

>>So my conclusion is that this doesn't belong in yotta itself. Does that make sense?
Not really. In practice developers will have random tools installed to do random things. Maybe they respect a clean PATH, maybe they don't. Do we really want to be in the middle of this? Its easier to operate in a sanitized environment IMO. Last thing I'd want to debug is a memory size problem just to find that path listed arm-none-aebi-gcc from a Si partner before the ARM_GCC release that I was expecting. Same goes for licensed versus eval tools. Its the wild west out there :dancer:

autopulated, 2015-04-02:
Hmm.

What if yotta respected a `YOTTA_SUBCOMMANDS_PATH` environment variable (which could also be specified in the yotta config file), which set the path up for its subprocesses.

If this variable isn't present, then yotta will just expose PATH to the subprocesses. Our installers can add things to `YOTTA_SUBCOMMANDS_PATH` when they install new tools (and any installers for targets could either set that environment variable or modify the yotta config file).

This would work across all platforms, not just windows.

sg-, 2015-09-28:
The windows installer implements this enhancement. Closing it here and will implement similarly for Mac and hopefully Ubuntu.

closed 2015-09-28 by sg-

BlackstoneEngineering, 2015-10-01:
It would be really nice if yotta did this by default, created a YOTTA_PATH variable that listed all its dependencies, then pre-pended this to the path. Currently we are relying on this on the installer side, but if it ever breaks then there is no easy fix. If it were a part of yotta then it would be fixed with an update by running `pip install -U yotta`

Also, would like to see a `YOTTA_Extras` path variable that the user can modify and have persist across multiple installations of yotta.

autopulated, 2015-10-02:
yotta **can't do that in the absence of an installer**, that's what I was trying to say: when you have an installer this makes sense, but if you don't yotta can't do it because it doesn't know where its dependencies are – it finds them through the path!