ARMmbed/yotta #212: No module named 'pwd' in windows

SeppoTakalo, 2015-04-07:
~~~
$ yt build                                                                             
Traceback (most recent call last):                                                     
  File "c:\Python27\Scripts\yt-script.py", line 9, in <module>                         
    load_entry_point('yotta==0.2.4', 'console_scripts', 'yt')()                        
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 546, in load_e  
ntry_point                                                                             
    return get_distribution(dist).load_entry_point(group, name)                        
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 2666, in load_  
entry_point                                                                            
    return ep.load()                                                                   
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 2339, in load   
    return self.resolve()                                                              
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 2345, in resol  
ve                                                                                     
    module = __import__(self.module_name, fromlist=['__name__'], level=0)              
  File "C:\Python27\lib\site-packages\yotta\__init__.py", line 6, in <module>          
    from .main import main                                                             
  File "C:\Python27\lib\site-packages\yotta\main.py", line 14, in <module>             
    from . import version                                                              
  File "C:\Python27\lib\site-packages\yotta\version.py", line 14, in <module>          
    from .lib import component                                                         
  File "C:\Python27\lib\site-packages\yotta\lib\component.py", line 15, in <module>    
    import access                                                                      
  File "C:\Python27\lib\site-packages\yotta\lib\access.py", line 13, in <module>       
    import target                                                                      
  File "C:\Python27\lib\site-packages\yotta\lib\target.py", line 19, in <module>       
    import pack                                                                        
  File "C:\Python27\lib\site-packages\yotta\lib\pack.py", line 28, in <module>         
    import vcs                                                                         
  File "C:\Python27\lib\site-packages\yotta\lib\vcs.py", line 15, in <module>          
    import fsutils                                                                     
  File "C:\Python27\lib\site-packages\yotta\lib\fsutils.py", line 8, in <module>       
    import pwd                                                                         
ImportError: No module named pwd                                                       
~~~

autopulated, 2015-04-07:
This is fixed on master. I'll publish a new release soon.

autopulated, 2015-04-27:
This was fixed by 9a041e458378edc3d943bee7f8674af56a6c8325

closed 2015-04-27 by autopulated