ARMmbed/yotta #213: Debug command fails with error: "TypeError: target_supported() takes exactly 3 arguments (2 given)"

SeppoTakalo, 2015-04-08:
~~~
$ yt debug source/mbed-coap-example
info: found source/mbed-coap-example
Traceback (most recent call last):
  File "c:\Python27\Scripts\valinor-script.py", line 9, in <module>
    load_entry_point('valinor==0.0.4', 'console_scripts', 'valinor')()
  File "c:\Python27\lib\site-packages\valinor\main.py", line 74, in main
    ide_tool = ide_detection.select(ide_detection.available(), args.target)
  File "c:\Python27\lib\site-packages\valinor\ide_detection.py", line 123, in select
    possible_ides = [x for x in available_ides if tool.target_supported(target, x)]
TypeError: target_supported() takes exactly 3 arguments (2 given)
Invalid parameter(s)
RESET { SESSION }
error: debug process exited with status 1
$ yt target
frdm-k64f-gcc,*
~~~

Platform: Windows 64bit

SeppoTakalo, 2015-04-08:
Updated to newest Yotta, same result
~~~
$ yt --version
0.2.5
~~~

0xc0170, 2015-04-08:
what version of valinor and pgen? Can you print pgen --version and valinor --version here?

0xc0170, 2015-04-08:
@SeppoTakalo For reference, currently yt using valinor tool, so this is connected to it as in the above code snippet (error is in valinor/ide_detection.py).

Can you test this PR : https://github.com/ARMmbed/valinor/pull/1 and if any issue (I have been testing uvision, You are using gcc gdb probably?) provide feedback in the pull request, where we can improve it before it gets merged. Would be appreciated. Thanks.

autopulated, 2015-04-08:
This is caused by breaking changes in project_generator 0.5, if you downgrade to 0.4.1, it should work:

```
pip uninstall project_generator
pip install project_generator==0.4.1
```

SeppoTakalo, 2015-04-08:
I can verify that downgrading the project_generator works.

autopulated, 2015-04-20:
This should be resolved now. A new version of valinor has been published that updates the project_generator dependency.

closed 2015-04-20 by autopulated