ARMmbed/yotta #214: Cryptography dependency error (windows)

0xc0170, 2015-04-08:
The output from invoking yt target, yotta installed from the latest master (commit f0c7b4caca04ff88ed85a972ae425)

```
 yt --version
0.2.5

yt target
Traceback (most recent call last):
  File "C:\Python27\Scripts\yt-script.py", line 5, in <module>
    from pkg_resources import load_entry_point
  File "C:\Python27\lib\site-packages\setuptools-14.3-py2.7.egg\pkg_resources\__init__.py", line 3023, in <module>
  File "C:\Python27\lib\site-packages\setuptools-14.3-py2.7.egg\pkg_resources\__init__.py", line 614, in _build_master
  File "C:\Python27\lib\site-packages\setuptools-14.3-py2.7.egg\pkg_resources\__init__.py", line 920, in require
  File "C:\Python27\lib\site-packages\setuptools-14.3-py2.7.egg\pkg_resources\__init__.py", line 807, in resolve
pkg_resources.DistributionNotFound: cryptography>=0.8
```

After running pip install cryptography, everything is fine.

autopulated, 2015-04-08:
Did you update from an existing installation, or do a clean install?

0xc0170, 2015-04-08:
Not certain what you are referring to. If I understood it correctly, I updated-  run python setup.py install from latest yotta master.

autopulated, 2015-04-08:
Ah OK. I don't believe this error happens if you install the released version with pip, so I'm going to close.

0xc0170, 2015-04-08:
OK, if will, reopen.

closed 2015-04-08 by 0xc0170