ARMmbed/yotta #215: Traceback in error message

bremoran, 2015-04-15:
When a module depends on a version in advance of what is available, yotta dumps a traceback.
```
Traceback (most recent call last):
  File "/usr/local/bin/yt", line 9, in <module>
    load_entry_point('yotta==0.1.0', 'console_scripts', 'yt')()
  File "/usr/local/lib/python2.7/site-packages/yotta/main.py", line 150, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/site-packages/yotta/update.py", line 28, in execCommand
    updateDeps(args)
  File "/usr/local/lib/python2.7/site-packages/yotta/update.py", line 51, in updateDeps
    available_components = [(c.getName(), c)]
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 447, in satisfyDependenciesRecursive
    provider = provider
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 300, in __getDependenciesRecursiveWithProvider
    _processed = _processed
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 300, in __getDependenciesRecursiveWithProvider
    _processed = _processed
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 300, in __getDependenciesRecursiveWithProvider
    _processed = _processed
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 300, in __getDependenciesRecursiveWithProvider
    _processed = _processed
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 300, in __getDependenciesRecursiveWithProvider
    _processed = _processed
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 282, in __getDependenciesRecursiveWithProvider
    provider = provider
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 190, in __getDependenciesWithProvider
    satisfyDep, specs
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 179, in satisfyDep
    update_installed
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 433, in provider
    r = access.satisfyVersionFromSearchPaths(name, version_req, search_dirs, update_if_installed)
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/access.py", line 253, in satisfyVersionFromSearchPaths
    v = latestSuitableVersion(name, version_required)
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/access.py", line 134, in latestSuitableVersion
    remote_component.spec
Exception: Github repository "<repo path>" does not provide a version matching ">=0.1.1,<0.2.0"
```

bogdanm, 2015-04-15:
Why is this a bad thing?

autopulated, 2015-04-15:
This is not an error in the implemntation of yotta, but an error that yotta has discovered in the modules it is installing – so the internal traceback is irrelevant, and a nice user-friendly error message should be displayed.

bogdanm, 2015-04-15:
Ah, sorry, I misread the original description. Yes, of course that makes sense.

autopulated, 2015-04-16:
This was fixed in 1bf1044b40019c8fc8f6ab7e8e854f3e8bc216a3

closed 2015-04-16 by autopulated