ARMmbed/yotta #219: Add tab completion to yotta

BlackstoneEngineering, 2015-04-18:
## Problem
There is no tab completion for yotta, to be a truely useful command line tool it need this. 

## Suggested tab completions
* target names
* common options
* anything that requires more than 3 characters typed 

## possible solution
https://argcomplete.readthedocs.org/en/latest/

autopulated, 2015-04-18:
argcomplete looks cool. Looks like it should be straightforward to integrate at a basic level, and shouldn't be too hard to add completions for things more complex things like `yotta target <tab>` after that.

pull-reqs welcome!

BlackstoneEngineering, 2015-04-18:
working on it in spare time, if anyone else manages to do it before me all the better!

autopulated, 2015-04-20:
Looks like your issues were caused by import order – seems argcomplete has to be before argparse.

I'm doing development on this at https://github.com/autopulated/yotta/tree/autocomplete if you're interested in following / helping out.

BlackstoneEngineering, 2015-04-20:
will do

BlackstoneEngineering, 2015-07-31:
Tab completion has been added to yotta, now looking to add support for hitting the yotta registry for completions.

mfiore02, 2015-09-27:
As a MBED developer at MultiTech, one of ARMs partner companies, I'd love to see tab completion built into yotta. That feature would make it a whole lot more useable.

autopulated, 2015-09-28:
Hi @mfiore02  – thanks for the feedback!

Happy to say that some support is already available: on Linux and Mac yotta supports argcomplete to complete subcommand names and options (unfortunately not module or target names yet though). To enable this, just activate argcomplete following the instructions [here](http://argcomplete.readthedocs.org/en/latest/), specifically:

```
pip install argcomplete
activate-global-python-argcomplete
```