ARMmbed/yotta #233: Verification with GitHub (account fails)

PrzemekWirkus, 2015-05-07:
# Description
I've tried to build project https://github.com/0xc0170/mbed-example-asynch-serial-private on my machine but GitHub verification failed badly.
After:
```
$ git clone https://github.com/0xc0170/mbed-example-asynch-serial-private
$ mbed-example-asynch-serial-private
$ yt target frdm-k64f-gcc
$ yt target
frdm-k64f-gcc,*
```

```
$ yt build
info: get versions for frdm-k64f-gcc
info: get versions for mbed
info: failed to fetch Github object, re-trying with authentication...

You need to log in with Github.
Press enter to continue.
Your browser will open to complete login.
```
Somewhere here I've opened webpage and I've used GitHub login button to login in.
After that yotta 'collapsed':
```
waiting for response...
If you are unable to use a browser on this machine, please copy and paste this URL into a browser:
http://yottabuild.org/?provider=github#login/c3NoLXJzYQAAAAMBAAEAAAEBALK5/o6tW6XiavBxELa3gOIGYLyOJCURqGoZr9XdN2eBx5bc1ADi3g/8K%2BdmSrwDjq6E15IV8RQEvUtxhys1a7dB3KHp2JEASTJn7YFv
x5gXccNaZqOGBmhEdEYQpfASOgtuXPeMNymxlICaQTI2hXLfMy7BZec20ptEPsK/SeNeTMh047v4yIXYi8iHInyWN0lJuovp4AmOSWoJQlxGzr6Rc%2Bssb2smzRrGah5XcvQPQjHPZltzZDb39KqhGxLys/nPzimAHh4O1N80cyGoT
2NXcV087GB0MeE64JzbIzUDbzG3cM816BhejbKvgWFHEVTnrrOd4mW3Rwsc9pn9F2M%3D
..Traceback (most recent call last):
  File "C:\Python27\Scripts\yt-script.py", line 9, in <module>
    load_entry_point('yotta==0.3.1', 'console_scripts', 'yt')()
  File "C:\Python27\lib\site-packages\yotta\main.py", line 171, in main
    status = args.command(args, following_args)
  File "C:\Python27\lib\site-packages\yotta\build.py", line 77, in execCommand
    install.execCommand(args, [])
  File "C:\Python27\lib\site-packages\yotta\install.py", line 58, in execCommand
    installDeps(args, c)
  File "C:\Python27\lib\site-packages\yotta\install.py", line 97, in installDeps
    test = {'own':'toplevel', 'all':True, 'none':False}[args.install_test_deps]
  File "C:\Python27\lib\site-packages\yotta\lib\component.py", line 508, in satisfyDependenciesRecursive
    test = test
  File "C:\Python27\lib\site-packages\yotta\lib\component.py", line 323, in __getDependenciesRecursiveWithProvider
    test = test
  File "C:\Python27\lib\site-packages\yotta\lib\component.py", line 223, in __getDependenciesWithProvider
    satisfyDep, specs
  File "C:\Python27\lib\site-packages\yotta\lib\component.py", line 209, in satisfyDep
    update_installed
  File "C:\Python27\lib\site-packages\yotta\lib\component.py", line 494, in provider
    r = access.satisfyVersionByInstalling(dspec.name, dspec.version_req, self.modulesPath())
  File "C:\Python27\lib\site-packages\yotta\lib\access.py", line 290, in satisfyVersionByInstalling
    v = latestSuitableVersion(name, version_required)
  File "C:\Python27\lib\site-packages\yotta\lib\access.py", line 122, in latestSuitableVersion
    vers = remote_component.availableVersions()
  File "C:\Python27\lib\site-packages\yotta\lib\github_access.py", line 262, in availableVersions
    for t in self._getTags():
  File "C:\Python27\lib\site-packages\yotta\lib\github_access.py", line 252, in _getTags
    self.tags = _getTags(self.repo).items()
  File "C:\Python27\lib\site-packages\yotta\lib\github_access.py", line 72, in wrapped
    authorizeUser()
  File "C:\Python27\lib\site-packages\yotta\lib\github_access.py", line 197, in authorizeUser
    if _pollForAuth(registry=registry):
  File "C:\Python27\lib\site-packages\yotta\lib\github_access.py", line 132, in _pollForAuth
    tokens = registry_access.getAuthData(registry=registry)
  File "C:\Python27\lib\site-packages\yotta\lib\registry_access.py", line 664, in getAuthData
    parsed_response = ordered_json.loads(body)
  File "C:\Python27\lib\site-packages\yotta\lib\ordered_json.py", line 29, in loads
    return json.loads(string, object_pairs_hook=OrderedDict)
  File "C:\Python27\lib\json\__init__.py", line 351, in loads
    return cls(encoding=encoding, **kw).decode(s)
  File "C:\Python27\lib\json\decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "C:\Python27\lib\json\decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
```

autopulated, 2015-05-07:
Please could you run the command that causes this error with `yotta -vvv <command>`, and paste the output into an email to me (do **not** paste it here, as it will contain secret access keys)

cc @headlessme: looks like a non-json page was returned by the registry when polling for the completed authentication. Not sure how this could happen.

PrzemekWirkus, 2015-05-07:
After deleting all dirs created by yotta, no change.

```
$ yt build
info: get versions for frdm-k64f-gcc
info: get versions for mbed
info: download mbed
info: get versions for mbed-cpputest
info: download mbed-cpputest
info: get versions for mbed-hal
info: get versions for cmsis-core
error: Failed to satisfy dependencies of c:\Work\martin\mbed-example-asynch-serial-private\yotta_modules\mbed:
error: mbed-hal does not exist in the modules registry. Check that the name is correct, and that it has been published.
error: cmsis-core does not exist in the modules registry. Check that the name is correct, and that it has been published.
error: mbed-hal not available
error: cmsis-core not available
error: Missing dependencies prevent build. Use `yotta ls` to list them.

$ yt ls
mbed-example-asynch-serial 0.0.4
|_ mbed 3.0.1
| |_ mbed-hal ~0.1.0 missing
| \_ cmsis-core ~0.1.0 missing
\_ mbed-cpputest 0.0.4
```

bogdanm, 2015-05-07:
Did you login to the private registry?

autopulated, 2015-05-07:
@PrzemekWirkus Please see my previous comment.

PrzemekWirkus, 2015-05-10:
1. Added private registry.
2. Can't reproduce this issue anymore (I can assume this was network error).

All things work well after step 1 & 2 :)

closed 2015-05-10 by PrzemekWirkus