ARMmbed/yotta #235: 'yotta install' does not install missing dependencies from linked modules

marcuschangarm, 2015-05-08:
Below, minar-platform is linked and depends on minar-platform-efm32 which isn't installed.

Yotta: 0.3.1 on Mac

```
mac:~/efm32/hello$ yt up
info: get versions for mbed
info: get versions for minar
info: get versions for swo
info: get versions for mbed-hal
info: get versions for cmsis-core
info: get versions for mbed-hal-silabs
info: get versions for mbed-hal-efm32gg
info: get versions for cmsis-core-silabs
info: get versions for cmsis-core-efm32gg
info: get versions for mbed-hal-efm32gg-watch
info: get versions for compiler-polyfill
info: get versions for swo-efm32

mac:~/efm32/hello$ yt ls
hello 0.0.0
┣─ mbed 3.0.1
┃ ┣─ mbed-hal 0.1.2 yotta_modules/mbed-hal
┃ ┃ ┗─ mbed-hal-silabs 0.1.2 yotta_modules/mbed-hal-silabs
┃ ┃   ┗─ mbed-hal-efm32gg 0.0.3 yotta_modules/mbed-hal-efm32gg
┃ ┃     ┗─ mbed-hal-efm32gg-watch 0.0.3 yotta_modules/mbed-hal-efm32gg-watch
┃ ┗─ cmsis-core 0.1.2 yotta_modules/cmsis-core
┃   ┗─ cmsis-core-silabs 0.1.2 yotta_modules/cmsis-core-silabs
┃     ┗─ cmsis-core-efm32gg 0.0.5 yotta_modules/cmsis-core-efm32gg
┣─ minar 0.0.1
┃ ┣─ compiler-polyfill 1.0.2 yotta_modules/compiler-polyfill
┃ ┗─ minar-platform 0.0.1 yotta_modules/minar-platform -> /Users/marcus/efm32/minar-platform
┃   ┗─ minar-platform-efm32 ARMmbed/minar-platform-efm32 missing
┗─ swo 0.1.0
  ┗─ swo-efm32 0.1.0 yotta_modules/swo-efm32

mac:~/efm32/hello$ yt install
mac:~/efm32/hello$ 
```

bogdanm, 2015-05-08:
That's because you linked your minar-platform module. yotta is expecting you to take care of the dependencies of the linked module by yourself.

marcuschangarm, 2015-05-08:
Is there a flag to tell yotta to do the package management?

bogdanm, 2015-05-08:
I think that's '-l'

```
$ yt install --help
usage: yt install [-h] [-l] [--test-dependencies {none,all,own}]
                  [--global | --save | --save-target]
                  [component]

Install dependencies for the current module, or a specific module.

positional arguments:
  component             If specified, install this module instead of installing the dependencies of the current module.

optional arguments:
  -h, --help            show this help message and exit
  -l, --install-linked  Traverse linked modules, and install dependencies needed there too.
  --test-dependencies {none,all,own}
                        Control the installation of dependencies necessary for building tests.
  --global, -g          Install globally instead of in the current working directory.
  --save                Add the specified module to dependencies in module.json
  --save-target         Add the specified module to targetDependencies in module.json
```

marcuschangarm, 2015-05-08:
Awesome! Thanks!

closed 2015-05-08 by marcuschangarm