ARMmbed/yotta #239: Need a way to invoke a test on a binary module.

bremoran, 2015-05-15:
CMake is breaking the yotta test paradigm in binary modules.

```
CMake Error at test/CMakeLists.txt:16 (target_link_libraries):
  Target "foo" of type EXECUTABLE may not be linked into
  another target.  One may link only to STATIC or SHARED libraries, or to
  executables with the ENABLE_EXPORTS property set.
```

It would be nice to be able to use yotta test against a binary module.

autopulated, 2015-05-15:
Unless it has undesirable side-effects, the fix for this is probably to set ENABLE_EXPORTS on yotta-defined executables.

autopulated, 2015-05-15:
This is done at https://github.com/autopulated/yotta/tree/fix-239

Note, this does not enable you to run executable modules as tests (through a testReporter script, etc), and at the moment I don't think we should do that.

autopulated, 2015-05-22:
fixed  in fa066361af32939c5f8529dafb315d5f9737626d

closed 2015-05-22 by autopulated