ARMmbed/yotta #244: yotta up fails depending on version of openssl installed (OSX)

bremoran, 2015-05-21:
Using yotta on mac OSX, I updated something in homebrew.  Following that update, yotta up yields this traceback:
```
Traceback (most recent call last):
  File "/usr/local/bin/yotta", line 9, in <module>
    load_entry_point('yotta==0.4.1', 'console_scripts', 'yotta')()
  File "/usr/local/lib/python2.7/site-packages/yotta/main.py", line 189, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/site-packages/yotta/list.py", line 39, in execCommand
    target, errors = c.satisfyTarget(args.target)
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 536, in satisfyTarget
    type = 'target'
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/access.py", line 352, in satisfyVersion
    return satisfyVersionByInstalling(name, version_required, working_directory, type=type)
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/access.py", line 298, in satisfyVersionByInstalling
    v = latestSuitableVersion(name, version_required, _registryNamespaceForType(type))
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/access.py", line 107, in latestSuitableVersion
    vers = remote_component.availableVersions()
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/registry_access.py", line 382, in availableVersions
    return _listVersions(self.namespace, self.name)
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/registry_access.py", line 166, in _listVersions
    request_headers = _headersForRegistry(registry)
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/registry_access.py", line 310, in _headersForRegistry
    auth_token = generate_jwt_token(_getPrivateKeyObject(registry), registry)
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/registry_access.py", line 299, in _getPrivateKeyObject
    privatekey_pem, None, default_backend()
  File "/usr/local/lib/python2.7/site-packages/cryptography/hazmat/backends/__init__.py", line 40, in default_backend
    _default_backend = MultiBackend(_available_backends())
  File "/usr/local/lib/python2.7/site-packages/cryptography/hazmat/backends/__init__.py", line 27, in _available_backends
    "cryptography.backends"
  File "build/bdist.macosx-10.9-x86_64/egg/pkg_resources/__init__.py", line 2351, in resolve
    
  File "/usr/local/lib/python2.7/site-packages/cryptography/hazmat/backends/openssl/__init__.py", line 7, in <module>
    from cryptography.hazmat.backends.openssl.backend import backend
  File "/usr/local/lib/python2.7/site-packages/cryptography/hazmat/backends/openssl/backend.py", line 1287, in <module>
    backend = Backend()
  File "/usr/local/lib/python2.7/site-packages/cryptography/hazmat/backends/openssl/backend.py", line 77, in __init__
    self._binding = Binding()
  File "/usr/local/lib/python2.7/site-packages/cryptography/hazmat/bindings/openssl/binding.py", line 114, in __init__
    self._ensure_ffi_initialized()
  File "/usr/local/lib/python2.7/site-packages/cryptography/hazmat/bindings/openssl/binding.py", line 126, in _ensure_ffi_initialized
    cls._modules,
  File "/usr/local/lib/python2.7/site-packages/cryptography/hazmat/bindings/utils.py", line 31, in load_library_for_binding
    lib = ffi.verifier.load_library()
  File "/usr/local/lib/python2.7/site-packages/cffi/verifier.py", line 97, in load_library
    return self._load_library()
  File "/usr/local/lib/python2.7/site-packages/cffi/verifier.py", line 207, in _load_library
    return self._vengine.load_library()
  File "/usr/local/lib/python2.7/site-packages/cffi/vengine_cpy.py", line 155, in load_library
    raise ffiplatform.VerificationError(error)
cffi.ffiplatform.VerificationError: importing '/usr/local/lib/python2.7/site-packages/cryptography/_Cryptography_cffi_e7d09016xc302a38b.so': dlopen(/usr/local/lib/python2.7/site-packages/cryptography/_Cryptography_cffi_e7d09016xc302a38b.so, 2): Symbol not found: _SSLv2_client_method
  Referenced from: /usr/local/lib/python2.7/site-packages/cryptography/_Cryptography_cffi_e7d09016xc302a38b.so
  Expected in: flat namespace
 in /usr/local/lib/python2.7/site-packages/cryptography/_Cryptography_cffi_e7d09016xc302a38b.so
```

The resolution for this problem seems to be installing cryptography (the python library) with extra environment variables:

```
env ARCHFLAGS="-arch x86_64" LDFLAGS="-L/usr/local/opt/openssl/lib" CFLAGS="-I/usr/local/opt/openssl/include" pip install cryptography
```

autopulated, 2015-05-21:
This seems to be caused by brew-installing a version of OpenSSL (and possibly brew linking it?), and then pip-installing cryptography with brew python.

autopulated, 2015-06-05:
Closing (not an issue within yotta). Hopefully this thread is google-able if anyone else has the same problem.

closed 2015-06-05 by autopulated

bremoran, 2015-09-15:
There is an update here:
```env LDFLAGS="-L$(brew --prefix openssl)/lib" CFLAGS="-I$(brew --prefix openssl)/include" pip install cryptography```

Here's the source: https://cryptography.io/en/latest/installation/