ARMmbed/yotta #245: Asm file did not Built by yotta.

eyeye, 2015-05-22:
I have a demo project.
two files in source dir: 
               main.c  ,  startup_stm32f401xe.s

target: 
               st-nucleo-f401re-gcc

build output:
**************************************************************************************************
[E:\yotta\demo0]$ yt build
warning: demo0 has invalid module.json:
warning:   homepage value u'' does not match u'^(https?://).*$'
info: generate for target: st-nucleo-f401re-gcc 0.0.8 at E:\yotta\demo0\yotta_targets\st-nucleo-f401re-gcc
GCC-C.cmake included
GCC-GXX.cmake included
GCC version is: 4.9.3
-- Configuring done
-- Generating done
-- Build files have been written to: E:/Yotta/demo0/build/st-nucleo-f401re-gcc
[1/2] Building C object source/CMakeFiles/demo0.dir/E_/Yotta/demo0/source/main.c.obj
[2/2] Linking C executable source/demo0
d:/progra~1/gnutoo~1/4bd65~1.920/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: warning: cannot find entry symbol Reset_Handler; defaulting to 08000000

**************************************************************************************************

in .build/st-nucleo-f401re-gcc/source dir CMakeLists.txt

**************************************************************************************************
include_directories("E:/yotta/demo0/source")

set(YOTTA_AUTO_DEMO0_C_FILES
    "E:/yotta/demo0/source/main.c"
)

add_executable (demo0
    ${YOTTA_AUTO_DEMO0_C_FILES}
)
**************************************************************************************************

How can I make the startup_stm32f401xe.s file to be built?

autopulated, 2015-05-22:
Thanks for reporting.

Auto-generating cmake lists for assembly files is not currently implemented. You can add a custom CMakeLists.txt file to the source directory as a workaround for this:

add_executable(demo0
    main.c 
    startup_stm32f401xe.s
)

Depending on your build target you may also have to add:

enable_language(ASM)

In the same file.

0xc0170, 2015-05-22:
@autopulated Is this planned to support?

autopulated, 2015-05-22:
Yeah, provided there are no unexpected complications, I think we should support this.

commit f02b9927 (2015-05-22): include .s assembly files in autogenerated CMakeLists.txt (fixes #245)
closed 2015-05-22 by autopulated