ARMmbed/yotta #253: Sending CTRL-C to GDB using 'yotta debug' frequently causes yotta and GDB to exit

marcuschangarm, 2015-05-28:
```
(gdb) c
Continuing.
^CTraceback (most recent call last):
  File "/usr/local/bin/yt", line 9, in <module>
    load_entry_point('yotta==0.4.1', 'console_scripts', 'yt')()
  File "/Library/Python/2.7/site-packages/yotta/main.py", line 189, in main
    status = args.command(args, following_args)
  File "/Library/Python/2.7/site-packages/yotta/debug.py", line 40, in execCommand
    error = target.debug(builddir, args.program)
  File "/Library/Python/2.7/site-packages/yotta/lib/target.py", line 374, in debug
    return self._debugDeprecated(builddir, program)
  File "/Library/Python/2.7/site-packages/yotta/lib/fsutils_posix.py", line 59, in wrapped_fn
    msg = q.get()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/queues.py", line 117, in get
    res = self._recv()
IOError: [Errno 4] Interrupted system call

Program received signal SIGTRAP, Trace/breakpoint trap.
```

autopulated, 2015-05-28:
Are you using the latest version 0.4.1? (It contained [this fix](https://github.com/ARMmbed/yotta/commit/7517bc9d334d17f44bc5984fef83e581e7c9f838))

autopulated, 2015-06-02:
Closing as wontfix, because this behaviour was identified with the now-deprecated `debug` and `debugServer` properties in target.json. Use `scripts.debug` instead.

closed 2015-06-02 by autopulated