ARMmbed/yotta #257: 'yt build' doesn't rebuild on header changes when building with ninja

bogdanm, 2015-06-05:
When building with `ninja`, changing a header included in one of the source files in the project doesn't cause the respective source file to be built (which is a real problem in Windows).
Reproducibility information:

```
$ git clone https://github.com/ARMmbed/yottabug.git
$ cd yottabug
$ yt target frdm-k64f-gcc
$ yt build (in Windows) or yt build -G Ninja (in Linux) # this will work
# now change RETVALUE in yottabug/header.h
$ yt build (in Windows) or yt build -G Ninja (in Linux) again
```

Result after the last step:

```
$ yt build -G Ninja
info: generate for target: frdm-k64f-gcc 0.0.16 at /home/bogdanm/work/mbed/temp/yottabug/yotta_targets/frdm-k64f-gcc
mbedOS.cmake included
GCC-C.cmake included
mbedOS-GNU-C.cmake included
GCC-GXX.cmake included
mbedOS-GNU-CXX.cmake included
GCC version is: 4.8.3
-- Configuring done
-- Generating done
-- Build files have been written to: /home/bogdanm/work/mbed/temp/yottabug/build/frdm-k64f-gcc
ninja: no work to do.
```

Looking a bit further at `build/frdm-k64f-gcc/build.ninja`, one can find this:

```
build source/CMakeFiles/yottabug.dir/home/bogdanm/work/mbed/temp/yottabug/source/yottabug.c.obj: C_COMPILER /home/bogdanm/work/mbed/temp/yottabug/source/yottabug.c
  DEFINES = -DYOTTA_COMPONENT_VERSION=\"0.0.0\"
  DEP_FILE = source/CMakeFiles/yottabug.dir/home/bogdanm/work/mbed/temp/yottabug/source/yottabug.c.obj.d
  FLAGS = -std=c99 -fno-exceptions -fno-unwind-tables -mcpu=cortex-m4 -mthumb -ffunction-sections -fdata-sections -Wall -Wextra -D__thumb2__ -Os -g -DNDEBUG -I/home/bogdanm/work/mbed/temp/yottabug -I/home/bogdanm/work/mbed/temp/yottabug/source    -include /home/bogdanm/work/mbed/temp/yottabug/build/frdm-k64f-gcc/yotta_config.h
  OBJECT_DIR = source/CMakeFiles/yottabug.dir
```

However, the `DEP_FILE` referenced above does not exist.
Not sure if this is actually a yotta bug or not, but it looks like a good place to start investigating.
Also, if make is used instead of ninja, this works as expected.

autopulated, 2015-06-05:
does this happen if you build to run natively on linux/mac? I think this might be a target description bug (or a target description exposing a CMake bug)

bogdanm, 2015-06-05:
Interesting. Unable to reproduce with `target-x86-linux-native`.

bogdanm, 2015-06-05:
Also, the difference is quite clear. With `target-x86-linux-native`:

```
/usr/bin/cc  -DYOTTA_COMPONENT_VERSION=\"0.0.0\" -std=c99 -O2 -g -DNDEBUG -I/home/bogdanm/work/mbed/temp/yottabug -I/home/bogdanm/work/mbed/temp/yottabug/source    -include /home/bogdanm/work/mbed/temp/yottabug/build/x86-linux-native/yotta_config.h -MMD -MT source/CMakeFiles/yottabug.dir/home/bogdanm/work/mbed/temp/yottabug/source/yottabug.c.o -MF "source/CMakeFiles/yottabug.dir/home/bogdanm/work/mbed/temp/yottabug/source/yottabug.c.o.d" -o source/CMakeFiles/yottabug.dir/home/bogdanm/work/mbed/temp/yottabug/source/yottabug.c.o   -c /home/bogdanm/work/mbed/temp/yottabug/source/yottabug.c
```

With `target-frdm-k64f-gcc`:

```
/home/bogdanm/bin/gcc-arm-none-eabi-4_8-2014q1/bin/arm-none-eabi-gcc  -DCPU_MK64FN1M0VMD12 -DTARGET_K64F -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -DMBED_OPERATORS -DYOTTA_COMPONENT_VERSION=\"0.0.0\" -std=c99 -fno-exceptions -fno-unwind-tables -mcpu=cortex-m4 -mthumb -ffunction-sections -fdata-sections -Wall -Wextra -D__thumb2__ -Os -g -DNDEBUG -I/home/bogdanm/work/mbed/temp/yottabug -I/home/bogdanm/work/mbed/temp/yottabug/source    -include /home/bogdanm/work/mbed/temp/yottabug/build/frdm-k64f-gcc/yotta_config.h -o source/CMakeFiles/yottabug.dir/home/bogdanm/work/mbed/temp/yottabug/source/yottabug.c.obj -c /home/bogdanm/work/mbed/temp/yottabug/source/yottabug.c
```

The K64F version doesn't have any options related to generating dependency files (.d). I have no idea why though.,

autopulated, 2015-06-05:
it's possible that something is missing from the target description that would describe how to do this – that raises the question of why it works with Makefiles though?

autopulated, 2015-06-05:
Oh, I notice a `.obj` extension there. That should be a `.o`.

Does this change if you update to the latest target description (earlier today)?

autopulated, 2015-06-05:
closing as we've identified that this is an issue with the target description used

closed 2015-06-05 by autopulated