ARMmbed/yotta #260: Yotta update failed when a specific version of dependency is specified

LiyouZhou, 2015-06-05:
Specified a exact version of dependencies for mbed-net-websocket, try to run yt update to get that version. Update failed. The following is the traceback.

```
liyzho01@e105699-lin:/work/iot/metrics-server$ yt update
warning: metrics-server has invalid module.json:
warning:   homepage value u'' does not match u'^(https?://).*$'
info: get versions for frdm-k64f-gcc
info: get versions for mbed
info: get versions for mbed-net-sockets
info: get versions for mbed-net-websocket
Traceback (most recent call last):
  File "/usr/local/bin/yt", line 9, in <module>
    load_entry_point('yotta==0.4.1', 'console_scripts', 'yt')()
  File "/usr/local/lib/python2.7/site-packages/yotta/main.py", line 189, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/site-packages/yotta/update.py", line 28, in execCommand
    updateDeps(args)
  File "/usr/local/lib/python2.7/site-packages/yotta/update.py", line 51, in updateDeps
    available_components = [(c.getName(), c)]
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 506, in satisfyDependenciesRecursive
    test = test
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 319, in __getDependenciesRecursiveWithProvider
    test = test
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 219, in __getDependenciesWithProvider
    satisfyDep, specs
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 205, in satisfyDep
    self.getName()
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 488, in provider
    r = access.satisfyVersionFromSearchPaths(dspec.name, dspec.version_req, search_dirs, update_if_installed)
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/access.py", line 268, in satisfyVersionFromSearchPaths
    v = latestSuitableVersion(name, version_required, registry=_registryNamespaceForType(type))
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/access.py", line 109, in latestSuitableVersion
    v = spec.select(vers)
  File "/usr/local/lib/python2.7/site-packages/semantic_version/base.py", line 480, in select
    options = list(self.filter(versions))
  File "/usr/local/lib/python2.7/site-packages/semantic_version/base.py", line 475, in filter
    if self.match(version):
  File "/usr/local/lib/python2.7/site-packages/semantic_version/base.py", line 470, in match
    return all(spec.match(version) for spec in self.specs)
  File "/usr/local/lib/python2.7/site-packages/semantic_version/base.py", line 470, in <genexpr>
    return all(spec.match(version) for spec in self.specs)
  File "/usr/local/lib/python2.7/site-packages/semantic_version/base.py", line 433, in match
    return version == self.spec
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/registry_access.py", line 345, in __eq__
    return self.namespace == other.namespace and \
AttributeError: 'Version' object has no attribute 'namespace'
```

autopulated, 2015-06-05:
I believe this is fixed on `master`, could you test that?

LiyouZhou, 2015-06-05:
New error on master: Now yt update fail outright 
liyzho01@e105699-lin:/work/iot/metrics-server$ yt update
warning: metrics-server has invalid module.json:
warning:   homepage value u'' does not match u'^(https?://).*$'
Traceback (most recent call last):
  File "/usr/local/bin/yt", line 9, in <module>
    load_entry_point('yotta==0.4.1', 'console_scripts', 'yt')()
  File "/usr/local/lib/python2.7/site-packages/yotta-0.4.1-py2.7.egg/yotta/main.py", line 189, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/site-packages/yotta-0.4.1-py2.7.egg/yotta/update.py", line 28, in execCommand
    updateDeps(args)
  File "/usr/local/lib/python2.7/site-packages/yotta-0.4.1-py2.7.egg/yotta/update.py", line 41, in updateDeps
    target, errors = c.satisfyTarget(args.target, update_installed=True)
  File "/usr/local/lib/python2.7/site-packages/yotta-0.4.1-py2.7.egg/yotta/lib/component.py", line 523, in satisfyTarget
    type = 'target'
  File "/usr/local/lib/python2.7/site-packages/yotta-0.4.1-py2.7.egg/yotta/lib/access.py", line 339, in satisfyVersion
    r = satisfyVersionFromSearchPaths(name, version_required, search_paths, update_installed == 'Update', type=type)
  File "/usr/local/lib/python2.7/site-packages/yotta-0.4.1-py2.7.egg/yotta/lib/access.py", line 255, in satisfyVersionFromSearchPaths
    v = latestSuitableVersion(name, spec, registry=_registryNamespaceForType(type))
  File "/usr/local/lib/python2.7/site-packages/yotta-0.4.1-py2.7.egg/yotta/lib/access.py", line 101, in latestSuitableVersion
    remote_component = remoteComponentFor(name, version_required, registry)
  File "/usr/local/lib/python2.7/site-packages/yotta-0.4.1-py2.7.egg/yotta/lib/access.py", line 62, in remoteComponentFor
    vs = sourceparse.parseSourceURL(version_required)
  File "/usr/local/lib/python2.7/site-packages/yotta-0.4.1-py2.7.egg/yotta/lib/sourceparse.py", line 53, in parseSourceURL
    parsed = urlsplit(source_url)
  File "/usr/local/lib/python2.7/urlparse.py", line 182, in urlsplit
    i = url.find(':')
AttributeError: 'Spec' object has no attribute 'find'

autopulated, 2015-06-06:
ack,
thanks for trying that

commit 0b83286d (2015-06-08): Merge pull request #264 from autopulated/master

fix #260
commit b0135b8d (2015-06-08): fix #260 (don't pass a Spec where a version requirement string is expected; add a test for this)
closed 2015-06-08 by autopulated