ARMmbed/yotta #267: Docs link is not visible from frontpage

bremoran, 2015-06-15:
It's not clear to new yotta users where to go to find more information about yotta or how to use it.  A documentation link on the front page might be helpful.

0xc0170, 2015-06-15:
What do you mean by the front page? Because readme references docs page https://github.com/ARMmbed/yotta#further-documentation , same is on pypi.

bremoran, 2015-06-15:
I mean that the [yotta landing page](http://yottabuild.org) makes no mention of docs.yottabuild.org.

danros, 2015-06-15:
Yotta will be explained on mbed.com. This will do things like explain briefly what yotta is, point to the docs site as well as point to the registry.

bremoran, 2015-06-15:
I fail to understand why that would prevent a link on yotta's homepage to its docs.  It shouldn't take a third-party webpage to find the documentation on what yotta is from its homepage.

I was under the impression that yotta was supposed to be a useful build system in general, not solely targeted at mbed.

0xc0170, 2015-06-15:
If yottabuild is the front page, I agree with @bremoran.
Look at npm, pypi, cargo and other package managers, there is present documentation link and many others useful links.

> It shouldn't take a third-party webpage to find the documentation on what yotta is from its homepage.

:+1:

autopulated, 2015-06-15:
The `yottabuild.org` isn't really "live", I don't know how anyone would land there, in fact?

autopulated, 2015-06-15:
(p.s. the website has a separate repository here: https://github.com/armmbed/yotta-website)

closed 2015-06-15 by autopulated