ARMmbed/yotta #273: mbed version defines for writing portable code

marcuschangarm, 2015-06-24:
Redirected from: https://github.com/ARMmbed/mbed-sdk-private/issues/27

Original post:
It would be useful to have defines for the major mbed releases, for example

```
#if defined(MBED2)
// do classic mbed stuff
#elif defined(MBEDOS)
// do mbed os stuff
#endif
```

autopulated, 2015-06-24:
Module version definitions have been needed for a while. How would this format work for you:

```
#define YOTTA_<modulename>_VERSION_STRING "1.2.3-whatever+stuffmightbehere"
#define YOTTA_<modulename>_VERSION_MAJOR 1
#define YOTTA_<modulename>_VERSION_MINOR 2
#define YOTTA_<modulename>_VERSION_PATCH 3
```

marcuschangarm, 2015-06-25:
Looks great! How would you generate the STRING variant?

autopulated, 2015-06-25:
From the full semantic version (including any pre-release or build metadata)

commit 072c8df8 (2015-06-26): provide version definitions, fixes #273

Both CMake set-definitions, and preprocessor definitions are defined:

```C
```

```
set(YOTTA_<MODULENAME>_VERSION_STRING "1.2.3-whatever")
set(YOTTA_<MODULENAME>_VERSION_MAJOR 1)
set(YOTTA_<MODULENAME>_VERSION_MINOR 1)
set(YOTTA_<MODULENAME>_VERSION_PATCH 3)
```
commit 0f4ea39d (2015-06-26): Merge pull request #275 from autopulated/version-defs

provide version definitions, fixes #273
closed 2015-06-26 by autopulated