ARMmbed/yotta #276: `yotta link` guide required

autopulated, 2015-06-25:
The docs site needs a guide to using `yotta link`, and `yotta link-target` adding. For people who don't have an npm background, these commands can be confusing.

sg-, 2015-06-26:
Feature I planned on adding but havent had much time is something like yotta --dev which pulls from the repo rather than the registry. Seems like *most* all the data is present to do this in module.json where the dependencies version could be scanned and pull the same code, just clone from the git release tag rather than pulling from the registry

```
  "repository": {
    "url": "git@github.com:mbedmicro/mbed_private.git",
    "type": "git"
  }
```

i'm sure you wont like this so wanted to propose before putting too much effort into it.

autopulated, 2015-06-26:
I don't think trying to pull things from repository arguments specified in the module descriptions is a good idea:

 * We shouldn't encourage people to edit things inside the yotta_modules cache. If you step in there you are treading into uncharted Here Be Dragons territory, where you may be surprised to find your changes overwritten or otherwise destroyed. For example, if you edit a module.json file so that it's invalid json then try to build, yotta will treat it as if the module didn't exist, and try to download it again overwriting any existing module.

 * If you have to link lots of modules to make a single bugfix / whatever, then those modules are too interdependent on each other. You should never normally have to link more than one module to make a pull request to fix something. (The *only* exception to this is adding an mbed-hal port, which you don't want anyone else to know about until it's completely finished (which prevents you from getting your target-specific dependencies included upstream before you're done, so you have to link both the hal implementation and the module that needs modifying to pull in your hal implementation for your target)).

* Even a small application will commonly have 20+ dependencies, each of which may be a git/hg repo with many MB of changes. Cloning all of those for development when you only needed one or two seems wasteful.

* Repository is not a required field in module.json, and it is not validated.

* The yotta_modules folder is not necessarily flat. (there may be nested `yotta_modules/moda/yotta_modules/modb/yotta_modules/etc` directories). So we might end up checkout out git repos inside existing git repos.

* `yotta link` works perfectly well for this, and puts you in complete control. It's a better solution all round.

sg-, 2015-06-26:
I'm not suggesting modifying the yotta module cache. Just suggesting a flag --dev that doesnt pull from the registry, rather uses metadata that exists in the module.json files to pull the development repo with history into the yotta_modules directory. This is non intrusive and the other method is adding a dev_dependencies section that only gets used when the --dev flag is passed. Lets look at the workflow proposed and I'd appreciate the steps using link to compare whats requested of the developers.
```
development> git clone http://github.com/armmbed/mbed-os-porting-program
development> cd mbed-os-porting-program
mbed-os-porting-program> yotta target gcc-port-target
mbed-os-porting-program>  yotta build --dev
```
At this point the directory yotta_modules within mbed-os-porting-program has a clone of our official documentation and shell (failing implementation of mbed hal, etc) and can start to develop and make changes. Next there would be a few options regarding commiting and uploading the changes. 
1. User goes into yotta_modules and edits the module.json for the port and add a new remote with git to push the repo to
2. We use a script similar to what mbed IDE currently does and traverse the yotta_modules directory and use yotta to edit the module_json and change the remote.
```
mbed-os-porting-program> cd yotta_modules
yotta_modules> ls
yotta_modules> cd some_module
some_module> git commit -am "My changes"
some_module> git add remote personal http://github.com/myaccount/some_module
some_module> git push personal master <or branch etc...>
```
Ideally we add this flow to yotta to find all the dependencies that were created but also dont think its unreasonable to use the above git workflow since I think that equivalent when using link (but there is room for improvement

autopulated, 2015-06-26:
the yotta_modules folder *is* the yotta modules cache

a devDependencies section is something that npm has, with a different meaning (for various reasons the equivalent yotta dependencies, which get installed only when you're compiling the module directly, not indirectly as a dependency, are called testDependencies instead)

sg-, 2015-06-27:
So when someone pulls dependencies with a program (guessing they always end up in yotta_modules) we're suggesting that they don't modify a dependency? I'm guessing that drivers for sensors will end up in there and if magic things can happen by the tool what type of development are people suppose to do? Sounds very scary to me. Just edit a main file? There are less that 10% (being generous) of libraries published on mbed without forks because they were written so well that no one has to modify when publishing their work based on it. Im getting nervous or hopefully misunderstand something...

autopulated, 2015-06-28:
This is the scary new world of modular software. Where you actually re-use code, instead of treating everything as an app note or example code from which to chisel away your own application.

The only circumstance where you modify a dependency (other than to update a new version) is to contribute a re-usable improvement or bugfix to it.

ansondtx20, 2015-06-28:
is it expected that every dependency will come with its own test harness that permits independent  checkout, test/fix, check in?

autopulated, 2015-06-28:
Yes, absolutely!

BlackstoneEngineering, 2015-07-31:
Question: would this be solved by modifying the sensor library module locally, then linking it to the executable module, also locally?

autopulated, 2015-11-16:
Fixed by #565

closed 2015-11-16 by autopulated