ARMmbed/yotta #278: support for prepublish scripts

autopulated, 2015-06-26:
Both modules and targets should be able to specify scripts that will be run by `yt publish`. These could add/verify license headers, or preprocess/generate/precompile resource files.

autopulated, 2015-06-26:
@pjbakker @mpg Would this work for your relicensing needs?

pjbakker, 2015-06-26:
Which needs specifically are you referring to?)

mbed TLS will be (after beta) licensed as Apache in source (just like everything else I hope).
We will need to publish a GPL package, but that won't be Yotta, I assume.

It could help for all components in general, as we will need the partner licensed version in the private registry. The question then is: Does it make sense to add those to the manifest, as it should only occur for the private registry, not our regular publications to the public registry.

Maybe I'm misunderstanding your purpose/goal here.

mpg, 2015-06-26:
The current plan is to have our own script to generate the yotta module anyway, so as far as I'm concerned it makes no big difference if relicensing happens when we generate the module or when we publish it.

(Besides, unless I'm mistaken the need to relicense should be temporary, once we've moved to Apache everyone will use the same license, right? @pjbakker?)

autopulated, 2015-06-26:
ok, thanks

PR #741 (2016-03-18): Scripting/hooks support
 * Adds `yotta start` aka `yotta run` defined by targets (defaulted to native execution).
 * Adds  prePublish, postPublish, postInstall, preGenerate, preBuild and postBuild module-defined scripts (plan to add also preTest, preDebug and preStart scripts)
 * Fixes #518
 * Fixes #278 
 * Fixes the use case in #611 (though not the request for arbitrary scripts)
 * Also some other bugfixes and improvements for the existing `test` and `debug` script support.
 * Fixes #234 (.py scripts can be used as testReporters, and their path is interpreted relative to the root of the modu [... cut]
closed 2016-03-18 by autopulated