ARMmbed/yotta #283: Yotta does not validate target.json before use

bremoran, 2015-06-30:
e.g. a missing toolchain.cmake generates the following output:
```
yt build
info: generate for target: target 0.0.0 at /path
Traceback (most recent call last):
  File "/usr/local/bin/yt", line 9, in <module>
    load_entry_point('yotta==0.4.2', 'console_scripts', 'yt')()
  File "/usr/local/lib/python2.7/site-packages/yotta-0.4.2-py2.7.egg/yotta/main.py", line 189, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/site-packages/yotta-0.4.2-py2.7.egg/yotta/build.py", line 101, in execCommand
    for error in generator.generateRecursive(c, all_components, builddir):
  File "/usr/local/lib/python2.7/site-packages/yotta-0.4.2-py2.7.egg/yotta/lib/cmakegen.py", line 110, in generateRecursive
    self.generate(builddir, modbuilddir, component, new_dependencies, dependencies, recursive_deps, toplevel)
  File "/usr/local/lib/python2.7/site-packages/yotta-0.4.2-py2.7.egg/yotta/lib/cmakegen.py", line 342, in generate
    "toolchain_files": reversed(self.target.getToolchainFiles())
  File "/usr/local/lib/python2.7/site-packages/yotta-0.4.2-py2.7.egg/yotta/lib/target.py", line 158, in getToolchainFiles
    os.path.join(x.path, x.description['toolchain']) for x in self.hierarchy
KeyError: 'toolchain'
```

autopulated, 2015-06-30:
yotta does check this file against a [schema](https://github.com/ARMmbed/yotta/blob/master/yotta/lib/schema/target.json), but the schema does not list this property as required (and it shouldn't be).

If this property isn't present, then it simply shouldn't be included in the CMake toolchain hierarchy.

bremoran, 2015-07-01:
Let me be more clear: when the "toolchain" property was not present in the json, the error above was generated.  This had nothing to do with whether there was a toolchain.cmake file present.

closed 2015-07-01 by bremoran

autopulated, 2015-07-01:
Right, but the toolchain property shouldn't have to be present!