ARMmbed/yotta #289: pip missing cython dependency

jacobrosenthal, 2015-07-02:
```bash
Jacobs-MacBook-Air-2:~ jacobrosenthal$ pip install yotta
Collecting yotta
  Using cached yotta-0.4.4.tar.gz
Requirement already satisfied (use --upgrade to upgrade): semantic-version<3,>=2.3.1 in /usr/local/lib/python2.7/site-packages (from yotta)
Requirement already satisfied (use --upgrade to upgrade): requests<3,>=2.4.3 in /usr/local/lib/python2.7/site-packages (from yotta)
Requirement already satisfied (use --upgrade to upgrade): PyGithub<2,>=1.25 in /usr/local/lib/python2.7/site-packages (from yotta)
Requirement already satisfied (use --upgrade to upgrade): colorama<0.4,>=0.3 in /usr/local/lib/python2.7/site-packages (from yotta)
Requirement already satisfied (use --upgrade to upgrade): hgapi<2,>=1.7 in /usr/local/lib/python2.7/site-packages (from yotta)
Requirement already satisfied (use --upgrade to upgrade): Jinja2<3,>=2.7.0 in /usr/local/lib/python2.7/site-packages (from yotta)
Collecting cryptography>=0.8 (from yotta)
  Using cached cryptography-0.9.1.tar.gz
Requirement already satisfied (use --upgrade to upgrade): PyJWT<2.0,>=1.0 in /usr/local/lib/python2.7/site-packages (from yotta)
Requirement already satisfied (use --upgrade to upgrade): pathlib<1.1,>=1.0.1 in /usr/local/lib/python2.7/site-packages (from yotta)
Requirement already satisfied (use --upgrade to upgrade): jsonschema<3.0,>=2.4.0 in /usr/local/lib/python2.7/site-packages (from yotta)
Collecting argcomplete<1.0,>=0.8.0 (from yotta)
  Using cached argcomplete-0.8.9-py2.py3-none-any.whl
Collecting mbed-test-wrapper<0.1.0,>=0.0.2 (from yotta)
  Using cached mbed_test_wrapper-0.0.2.tar.gz
Collecting valinor<1.0,>=0.0.0 (from yotta)
  Using cached valinor-0.0.6.tar.gz
Requirement already satisfied (use --upgrade to upgrade): markupsafe in /usr/local/lib/python2.7/site-packages (from Jinja2<3,>=2.7.0->yotta)
Collecting idna (from cryptography>=0.8->yotta)
  Using cached idna-2.0-py2.py3-none-any.whl
Collecting pyasn1 (from cryptography>=0.8->yotta)
  Using cached pyasn1-0.1.8.tar.gz
Collecting six>=1.4.1 (from cryptography>=0.8->yotta)
  Using cached six-1.9.0-py2.py3-none-any.whl
Requirement already satisfied (use --upgrade to upgrade): setuptools in /usr/local/lib/python2.7/site-packages/setuptools-12.0.5-py2.7.egg (from cryptography>=0.8->yotta)
Collecting enum34 (from cryptography>=0.8->yotta)
  Using cached enum34-1.0.4.tar.gz
Collecting ipaddress (from cryptography>=0.8->yotta)
  Using cached ipaddress-1.0.7-py27-none-any.whl
Collecting cffi>=0.8 (from cryptography>=0.8->yotta)
  Using cached cffi-1.1.2.tar.gz
Requirement already satisfied (use --upgrade to upgrade): functools32 in /usr/local/lib/python2.7/site-packages (from jsonschema<3.0,>=2.4.0->yotta)
Collecting pyyaml<4,>=3 (from valinor<1.0,>=0.0.0->yotta)
  Using cached PyYAML-3.11.tar.gz
Collecting pyOCD<1.0,>=0.3 (from valinor<1.0,>=0.0.0->yotta)
  Using cached pyOCD-0.4.4.zip
Collecting project-generator<1.0,>=0.5.7 (from valinor<1.0,>=0.0.0->yotta)
  Using cached project_generator-0.6.1.zip
Collecting pycparser (from cffi>=0.8->cryptography>=0.8->yotta)
  Using cached pycparser-2.14.tar.gz
Collecting intelhex (from pyOCD<1.0,>=0.3->valinor<1.0,>=0.0.0->yotta)
  Using cached intelhex-2.0-py2.py3-none-any.whl
Collecting hidapi (from pyOCD<1.0,>=0.3->valinor<1.0,>=0.0.0->yotta)
  Using cached hidapi-0.7.99-6.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 20, in <module>
      File "/private/var/folders/vx/8zp3pb4n6bj1_n6jwnfb3hf80000gn/T/pip-build-uZRnR1/hidapi/setup.py", line 4, in <module>
        from Cython.Distutils import build_ext
    ImportError: No module named Cython.Distutils
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/vx/8zp3pb4n6bj1_n6jwnfb3hf80000gn/T/pip-build-uZRnR1/hidapi
Jacobs-MacBook-Air-2:~ jacobrosenthal$ python
Python 2.7.9 (default, Feb  6 2015, 14:00:21) 
[GCC 4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.56)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from Cython.Distutils import build_ext
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named Cython.Distutils
>>> exit()
```

0xc0170, 2015-07-02:
This is an issue for pyOCD, reported here https://github.com/mbedmicro/pyOCD/issues/163

autopulated, 2015-07-02:
Closing per @0xc0170's comment, please track https://github.com/mbedmicro/pyOCD/issues/163

closed 2015-07-02 by autopulated