ARMmbed/yotta #293: contents of yotta_modules should be read-only

bremoran, 2015-07-03:
If the contents of yotta_modules are touched during development, it can cause false positives in test results prior to publishing.  By setting yotta_modules as read-only, it encourages users to use yotta link instead of modifying yotta_modules.

BlackstoneEngineering, 2015-07-31:
Would it not be better to simply hash to yotta modules folder and check it when you're building, then issue a big red warning message to the user to not muck around in the modules folder, but instead do X, where X is a link to a guide on how to use yotta link and develop modules and executables?

bridadan, 2015-07-31:
If I was user, I would prefer the warning over forcing read-only. The reason is say I'm not familiar with yotta, npm, or any kind of package manager. I just want to make a change in a module. The first thing I'm going to try is edit the module in the `yotta_modules` directory because I don't know any better. If the yotta_modules directory is read-only I would probably just be confused. But if i make my change, build, and get a "big red warning" message, I'm going to be educated on how make changes correctly. And I would at least be able to make tiny experimental changes while understanding that these changes can't be published.

autopulated, 2015-07-31:
Yeah that sounds like a much nicer experience.