ARMmbed/yotta #297: [minor] `yotta init` to produces old-style `license` property in `module.json`

hugovincent, 2015-07-06:
As title. 

Yotta init produces a module.json with e.g. `  "license": "Apache-2.0",` instead of the form shown in the docs:
```json
 "licenses": [
    {
      "url": "https://spdx.org/licenses/Apache-2.0",
      "type": "Apache-2.0"
    }
  ],
```

The short-name-only form is valid against the schema, but it is confusing that it is different than the docs, and we want to encourage people to include the SPDX URLs for their license choices wherever possible.

autopulated, 2015-07-06:
"license": "<SPDX-ID>" is a supported syntax. You can derive the full URL from that anyway.

In the future I would like to move to full SPDX license expression syntax (which actually requires the use of the short single license: property, but allows you to express multiple conjunctive/disjunctive licenses within that, see here, for example: https://docs.npmjs.com/files/package.json#license

The docs should document that "license" is supported though.

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-902

autopulated, 2015-11-16:
Closing as invalid: the `license` field will be preferred in the future, support for license expression syntax is listed in a separate issue.

closed 2015-11-16 by autopulated