ARMmbed/yotta #298: yotta init should fill in a default .gitignore if run in a git repo.

bremoran, 2015-07-06:
All yotta modules require similar entries in .gitignore.  Adding this functionality to yotta init would save a repetitive task.

bogdanm, 2015-07-06:
+1. Remember to also put `.yotta.json` there.

BlackstoneEngineering, 2015-07-31:
Why stop there, add a mercurial ignore file too, heck maybe add them both regardless of where the repo is initialized.

autopulated, 2015-08-03:
I don't think yotta should generate files that aren't for an existing git repo.

BlackstoneEngineering, 2015-08-03:
I will often create a module locally and test before I ever create a git repo for it. In fact this is how i develop most often. In this workflow having the .git ignore files present would be rather helpful.

## Workflow
Local yotta module-> test -> modify-> repeat -> commit to git

thegecko, 2015-08-03:
I'm not sure this is the responsibility of yotta.
But this can definitely be considered in IDEs ;)