ARMmbed/yotta #300: Check all logging doesn't output to stderr

thegecko, 2015-07-07:
info: and debug: logging should go to stdout
warning: and error: logging should go to stderr

Currently all yotta output could go to stderr.

This bug was seen due to yotta --version going to stderr, but this seems to be a python issue:

https://bugs.python.org/issue18920

We could remove argparse for --version to fix that problem

autopulated, 2015-07-08:
`--version` now goes to stdout, but I'll leave this open as info/error logging still needs splitting

geky, 2016-03-02:
+1, I think this might be the same issue

yt version (not --version) is also going to stderr

thegecko, 2016-03-02:
I think this is a regression since 0.14.0 or 0.14.1

thegecko, 2016-03-10:
Build output is also now outputting to stderr, another regression.