ARMmbed/yotta #309: module initialisation functions

autopulated, 2015-07-13:
yotta should provide a way for each module to have an initialisation function called.

preferred method:
 * `__yotta_module_init` or `__yotta_module_<modulename>_init` which you add  __attribute__((constructor)) to, (toolchains possibly provide a nice abstraction for this attribute). when creating your library yotta adds the right -Wl,-u etc. options to pull in this symbol if it's undefined, and to suppress errors if the symbol isn't defined.

I've been experimenting, but haven't figured out if this is uniformly possibly yet.

If the preferred method isn't possible, other implementations might include:
 * add method name(s) to module.json that you must have defined
 * instead of using the constructor attribute and c library startup code to run the init function, a yotta-init module could collect them all with custom CMake, and provide a idempotent initialisation API that any module can call to ensure initialisers have been called.
 * system libraries (like mbed) that define startup code may hide the use of this initialisation library from the user by calling it at startup.

Preferably whatever mechanism is used to call the init function should support calling de-init functions (at least in theory), so that the possibility of dynamically loading and unloading yotta modules is not precluded.

hugovincent, 2015-07-13:
> I've been experimenting, but haven't figured out if this is uniformly possibly yet.

Do you mean across toolchains? Looks like at least gcc, Clang and [ARM tools support this gcc extension](http://infocenter.arm.com/help/topic/com.arm.doc.dui0376d/CJADJIGE.html) too, which is good.

Is `__yotta_module_init ` (or `__yotta_module_<modulename>_init`) a name Yotta is explicitly aware of?  Or is it something used by convention (with the only explicit awareness on Yotta being the `-u __yotta_module_init` etc)? For user-friendliness and search-ability (and to hide any toolchain variations), maybe Yotta should define `YOTTA_MODULE_INIT()` or something at the compiler command line that expands to e.g. `void __yotta_module_<modulename>_init() __attribute__((constructor))`. Then in the module source, you would write:
```C
YOTTA_MODULE_INIT()
{
   ...
}
```

(I wonder if we should also document the existence and use of `__attribute__((destructor))` (even if it won't ever be called at present, at least in mbed) and/or provide the appropriate macro as above to better support dynamic loading/unloading in the future?)

From reading [this](http://stackoverflow.com/questions/2053029/how-exactly-does-attribute-constructor-work), it is good to know that priorities (to control ordering) are available if we need them, but choosing the priorities would be pretty difficult for a module author. Could Yotta do anything to help? For example if a macro was defined as above, maybe priorities could be synthesised from the Yotta depgraph. Or at least give us the option to do that in the future.

Note that either relying on C runtime, or an idempotent API as you describe may prevent or complicate some more advanced future use cases for this in early boot in mbed, where more explicit ordering control may be needed (example, with reference to our conversation on Slack: an embedded platform with DRAM on a SODIMM, using an I2C/SMBus memory to read the [EEPROM config data](https://en.wikipedia.org/wiki/Serial_presence_detect) before configuring the memory and allocator). To solve this, the simplest way would probably be to introduce groups, and provide the guarantee that all initialisers in one group are called before the next group are called -- that could probably be done with priorities, a macro, and a contract across the system for how groups are used. But having explicit named groups collected into named sections would permit more control and be more readable and understandable.

hugovincent, 2015-07-13:
My preference, even if it's a little less generic and the implementation more complex, is to permit yotta to collect specified sets of initialisation symbols matching a specified pattern. Example semantics:
* permit arbitrarily many "exports" specified in module.json using e.g. an "exports" key, that by default do nothing.
* an ability to capture them into groups using (JSON-schema flavoured) regexes that operate over all exports the exports in a given module's dependencies.
* have Yotta generate header file(s) for them. Example top-level `module.json`:
```JS
{
  "name": "helloyotta",
  ...
  "captureExportsGroup": {
    "pattern": "^(ExampleGroup_.*)$",
    "transform": "&$1", // this would use Javascript "".replace(/regex/, "$1") semantics for example.
    "join": ",\n",
    "header": "fptr_t myInitFunctionPointers = {\n",
    "footer": "}\n",
    "generatedFile": "relative/path/to/MyInitFunctions.h"
  }
}
```
Resulting generated C header file:
```C
fptr_t myInitFunctionPointers = {
    &ExampleGroup_example1,
    &ExampleGroup_example2
}
```
Big disadvantage of this include the fragility of non-trivial JavaScript regexes and transforms when implemented in Python, and that it's making Yotta be a bit of a programming language which might be undesirable.

autopulated, 2015-07-14:
> Do you mean across toolchains? Looks like at least gcc, Clang and ARM tools support this gcc extension too, which is good.

Yes, but this isn't the hard part: the problem is being able to ask the linker to pull in a symbol whether or not you've actually defined it.

> Is __yotta_module_init (or __yotta_module_<modulename>_init) a name Yotta is explicitly aware of? Or is it something used by convention (with the only explicit awareness on Yotta being the -u __yotta_module_init etc)? 

Yes, it's explicitly aware of it because it has to do `-u` or whatever.

> YOTTA_MODULE_INIT

Sure, why not.

> (I wonder if we should also document the existence and use of __attribute__((destructor)) (even if it won't ever be called at present, at least in mbed) and/or provide the appropriate macro as above to better support dynamic loading/unloading in the future?)

Yup, that's what I was thinking of as the corresponding unload mechanism.

> From reading this, it is good to know that priorities (to control ordering) are available if we need them, but choosing the priorities would be pretty difficult for a module author. Could Yotta do [...]

No, as we discussed, without complete coordination between all module authors, or the app author defining everything, init functions will have to be written in a very defensive manner to be safe. I don't believe there's any way to avoid that.

> Note that either relying on C runtime, or an idempotent API as you describe may prevent or complicate some more advanced future use cases for this in early boot in mbed, where more explicit ordering control may be needed (example, with reference to our conversation on Slack: an embedded platform with DRAM on a SODIMM, using an I2C/SMBus memory to read the EEPROM config data before configuring the memory, before enabling the memory allocator. Thus some early boot stuff might have to be runnable without the ability to malloc).

Defensively programming init functions wouldn't necessarily require mallocing. You'd just need to use an EEPROM driver that didn't need to malloc.

> My preference, even if it's a little less generic and the implementation more complex.

Umm, OK – why would you prefer something less generic and more complex over something simpler and more generic?

The most important goal of this is to do it in a way that is as simple as possible – for people to use, if not for us to implement!

autopulated, 2015-07-14:
so:

associating the correct undefined flags with a static library so that they get propagated by cmake to any executable that links that library has proven impossible to do in a straightforward way.

I'm now thinking that having an "modloader" module that provides an API for the system startup code to call all the registered initialisation functions is the way to go.

This module will probably work by generating source from CMake based on a yotta-defined list of exported symbols: in any case there will be nothing special about it, so replacing/extending it with functionality to provide ordering/groups of init functions should be possible. Multiple such modules should even be able to co-exist in an application.

hugovincent, 2015-07-14:
If the undefined flags thing didn't work, why not have a module.json key to tell the build system that you're using `YOTTA_MODULE_INIT`? If you define it as true and the symbol is missing, that is correctly an error.

hugovincent, 2015-07-14:
I still prefer my proposed approach above with module-controlled export and collection preferable to the "modloader" approach.

autopulated, 2015-07-14:
To be clear. It's possible to make undefining symbols _work_, but what I can't make work in a sane way is for you to be able to link against a module in custom CMake, and have its initialisation function called correctly – without significant additional yotta magic in all custom CMakeLists.

It isn't so much the "telling the build system that you're using an init function" that's the problem, but propagating the flags to pull in that init function through CMake to everywhere they need to be.

It isn't possible to just add an extra  `target_link_libraries(your module "-Wl,-u,__yotta_module_load" "-Wl,-u,__yotta_module_unload")` because those flags will appear *after* your library on the linker command line of any executable that uses your library – they need to appear before it. As far as I can find there is no built-in mechanism for doing that (and simply listing the library in its own dependencies so that it appears twice doesn't work either, as CMake deduplicates it).

autopulated, 2015-07-14:
(I didn't appreciate this difficulty when I wrote my first comment, which is probably the cause of confusion)

hugovincent, 2015-07-14:
Thanks for the clarification.

hugovincent, 2015-07-14:
After more discussion, I think we agree that Yotta shouldn't itself do the string munging as in my proposal above. How about instead, the Yotta module.json has a key for exporting symbols that are made available to the build system, so that the munging can be done by a script called by CMake?

bremoran, 2015-10-16:
I'm wondering if the right answer here is to generate a C/C++ file that calls the functions defined in module init.

bremoran, 2015-10-23:
As demonstrated in https://github.com/ARMmbed/mbed-example-network/issues/31, the lack of module initialization functions forces network objects to be created on the heap.  It is not possible to statically allocate a network object, since it must be initialized after the stack registration (e.g. ```lwipv4_init()```).

autopulated, 2015-10-23:
@bremoran got it, thanks.

bogdanm, 2015-10-23:
It feels a bit weird and dangerous to me to rely on yotta to define initialization order for C/C++ functions and/or objects. If we do it, at least we need to let the users know how their initialization works.

bogdanm, 2015-10-23:
Ok, fair enough. But we also need to be prepaired for situations when a module needs to initialize something before another module on which it depends, because this might happen sooner or later. Also, it A depends directly on B, C and D, what should be order for calling B, C and D's respective initialization functions? (I'm guessing the same order in which they are listed in module.json?)

bremoran, 2015-10-23:
The order should be irrelevant, otherwise the modules should express a dependency.