ARMmbed/yotta #313: Symlinked module being replaced by one of modules without any message

0xc0170, 2015-07-16:
It got into weird compilation errors where I symlinked many modules which resulted in errors. Yt ls command does not indicate much, only that one of symlinked modules is not symlinked (no green color). I check yotta_modules, the module is in there symlinked. But one of symlinks is ignored and another module downloads newer version to its yotta_modules, this one gets in, not my symlinked repository.

| |       \_ mbed-hal-frdm-k64f 0.2.0 yotta_modules\mbed-hal\yotta_modules\mbed-hal-frdm-k64f

I symlinked mbed-hal-frdm-k64f version 0.1.2. The only hint is yt ls -a red numbers that some dependencies are not matched, and that the module in a list is not shown as symlinked.  Once I do rebase my module to get the latest changes and resolve conflicts,  I am good to go.

It took some time to get through error messages to realize which modules are not getting in, start with red numbers (got few) ended up with matching module json files to see why I am not up to date.

A message that symlinked module does not meet requirements by xxx module with version x.x.x ?

autopulated, 2015-07-16:
version 0.5.0 should have fixed the case where `yt ls` would fail to show a version error for linked modules – if you have a case you can share where that still happens that'd be really helpful!

autopulated, 2015-07-16:
After further discussion:

The issue here appears to be that the linked module (at the top level) didn't satisfy a version constraint of the module that was using it, so a (non-linked) version *already installed on disk* in the module's own yotta_modules folder was used instead.

One solution to this would be to change the lookup rules so that linked modules are always used, even if they do not match version constraints (currently they will always be used only if no other modules are found in the search path that satisfy the constraints).

The visibility of version constraints in the `yotta ls -a` output could also be improved (they are currently omitted for the first user of each module).

autopulated, 2015-08-06:
closing as this isn't a bug as described

closed 2015-08-06 by autopulated