ARMmbed/yotta #318: yotta doesn't update local dependencies of linked modules

bremoran, 2015-07-20:
If the following conditions are true...
* module A depends on module B
* module B depends on module C
* module B is linked by module A
* module C is not present in module B's yotta modules

Then the following two bugs seems to appear:
* module C is silently installed in module A's yotta_modules
* ```yotta update```, executed in module A's directory, will not update module C

autopulated, 2015-07-21:
I edited your description to try to clarify this, can you confirm that's accurate?

bremoran, 2015-07-22:
Yes, this is accurate.

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-901

tabarr, 2015-10-29:
I am seeing the same issue. I had to make some TARGET define changes to sal-stack-lwip for a new target I am working on that is K64F "like". So I moved sal-stack-lwip to my sandbox and created links in my project. When I ran "yt update" initially I was getting multiple version mismatches, which I fixed by deleting the modules and running "yt update" again, except for one module. This module is like module C in the OP's example above. Here is the "yt ls" output for my project:

(workspace) c:\mbedos\projects\mbed-client-example-6lowpan>yt ls
mbed-client-example-6lowpan 0.0.2
|_ mbed-client 1.1.15
| |_ mbed-client-c 1.1.1 yotta_modules\mbed-client-c
| | \_ nanostack-libservice 3.0.7 yotta_modules\nanostack-libservice
| |_ mbed-client-mbed-os 1.0.11 yotta_modules\mbed-client-mbed-os
| | |_ mbed-drivers 0.11.1 yotta_modules\mbed-drivers
| | | |_ mbed-hal 1.0.3 yotta_modules\mbed-hal
| | | | \_ mbed-hal-freescale 1.0.0 yotta_modules\mbed-hal-freescale
| | | |   \_ mbed-hal-ksdk-mcu 1.0.3 yotta_modules\mbed-hal-ksdk-mcu
| | | |     |_ uvisor-lib 1.0.4 yotta_modules\uvisor-lib
| | | |     \_ mbed-hal-k64f 1.0.1 yotta_modules\mbed-hal-k64f
| | | |       \_ mbed-hal-frdm-k64f 1.0.0 yotta_modules\mbed-hal-frdm-k64f
| | | |_ cmsis-core 1.0.1 yotta_modules\cmsis-core
| | | | \_ cmsis-core-freescale 1.0.0 yotta_modules\cmsis-core-freescale
| | | |   \_ cmsis-core-k64f 1.0.0 yotta_modules\cmsis-core-k64f
| | | |_ ualloc 1.0.2 yotta_modules\ualloc
| | | | \_ dlmalloc 1.0.0 yotta_modules\dlmalloc
| | | |_ minar 1.0.1 yotta_modules\minar
| | | | \_ minar-platform 1.0.0 yotta_modules\minar-platform
| | | |   \_ minar-platform-mbed 1.0.0 yotta_modules\minar-platform-mbed
| | | |_ core-util 1.0.1 yotta_modules\core-util
| | | \_ compiler-polyfill 1.1.1 yotta_modules\compiler-polyfill
| | \_ sockets 1.0.2 yotta_modules\sockets
| |   \_ sal 1.0.2 yotta_modules\sal
| |     \_ sal-stack-lwip 1.0.1 yotta_modules\sal-stack-lwip -> c:\mbedos\dev-modules\sal-stack-lwip  <<< LINKED MODULE
| |       |_ sal-driver-lwip-k64f-eth ^1.0.0 missing   <<<<< MODULE NOT DOWNLOADING
| |       \_ sal-iface-eth 1.0.0 yotta_modules\sal-iface-eth -> c:\mbedos\dev-modules\sal-iface-eth
| \_ mbed-client-mbedtls 1.0.6 yotta_modules\mbed-client-mbedtls
|   \_ mbedtls 2.2.0-rc.1 yotta_modules\mbedtls
|_ atmel-rf-driver 1.0.1 >=0.0.17,<0.1.0 -> C:\mbedos\dev-modules\atmel-rf-driver
| \_ sal-stack-nanostack 3.0.5 yotta_modules\sal-stack-nanostack
|   |_ sal-stack-nanostack-eventloop 1.0.3 yotta_modules\sal-stack-nanostack-eventloop
|   | \_ mbed-6lowpan-eventloop-adaptor 1.0.2 yotta_modules\mbed-6lowpan-eventloop-adaptor
|   \_ nanostack-randlib 1.0.0 yotta_modules\nanostack-randlib
\_ mbed-mesh-api 0.3.6
  \_ sal-iface-6lowpan 0.2.15 yotta_modules\sal-iface-6lowpan

autopulated, 2015-10-30:
@tabarr thanks for reporting: to work around it if you install sal-driver-lwip-k64f-eth at the top-level it should work correctly.