ARMmbed/yotta #319: yotta link & support for multi-user systems

bremoran, 2015-07-22:
A useful feature would be a way to explicitly specify the target of a yotta link.

Imagining yotta in use on a build farm, it's easy to see that linking modules becomes impractical.  Even if yotta supports user-private links, it's easy to imagine the build user being the same across many parallel builds.

A potential solution to this problem is something like:
```
yotta link <module name> --explicit <path to linked module>
```

autopulated, 2015-07-22:
Yeah, maybe – it might get confusing to have links that work in two different ways though?

Note that the global link directory will be placed under YOTTA_PREFIX if that env var is set.

You could also create symlinks manually if you are using yotta in an automated environment.

bremoran, 2015-07-22:
Yes, creating symlinks manually is the default.  The point of doing it through yotta is that it could resolve the module name and create the correct name automatically.
```
yotta link --explicit <path to linked module>
```

0xc0170, 2015-07-22:
I like the last idea link --explicit. I used to have two clones of some repo (private suffix, remember?), and had to go back to the right module, link and go back to link the correct one.

AlessandroA, 2015-07-22:
If we managed to use `exec_prefix` also on *nix we could use `virtualenv`:

```bash
$ mkdir ~/dev_env
$ virtualenv ~/dev_env
$ cd ~/dev_env && source bin/activate
(dev_env) $ pip install yotta
...
(dev_env) $ git clone yt_module && cd yt_module
(dev_env) $ yt link
```
So the link only exists in the virtual environment.

The problem is that on some *nix `exec_prefix` is `/usr/`. We could do something like
```python
if os.name == 'nt':
    return sys.exec_prefix
else:
    if sys.exec_prefix == "/usr":
        return '/usr/local'
    else:
        return sys.exec_prefix
```
(not very elegant)

bremoran, 2015-07-22:
@AlessandroA IMO, that's an awfully big hammer for an awfully small problem...

AlessandroA, 2015-07-22:
well, it does not only solve that small problem. with `virtualenv` you get the benefit of full "clean" testability. you can even test with different versions of the tools/dependencies

bridadan, 2015-07-22:
I do like the `virtualenv` idea. That way we're not adding continuous-integration-specific tools into yotta if we can help it. `virtualenv` is used on most CI systems anyway so it's fairly standard practice. Plus, as @AlessandroA mentioned, it allows you test different versions of tools right out of the box.