ARMmbed/yotta #320: Yotta to stash linked modules

0xc0170, 2015-07-22:
There were many times problems during the development, when clean of modules was required to do to get fresh new dependencies (not all of them were linked), which caused that link modules were gone. what's the behaviour for linked modules after yotta clean, or when changing a target?

I overcome this taking pictures of all my linked modules, and manually restore them. Is there a way how yotta can capture what are linked repo (at least names for give target)? I am thinking about something like stash pop/push functionality.

@rgrover You had similar problems when local linking, you might add some more info.

autopulated, 2015-07-22:
to be clear: `yotta clean` does not remove the modules cache directory, only the build directory.

It sounds like we should fix whatever the underlying problem was that caused you to have to remove the modules directory in the first place: it might have been [this issue](https://github.com/ARMmbed/yotta/issues/318)?

0xc0170, 2015-07-22:
I have been testing the latest yotta version in the last 2 hours, and seems to be consistent when changing targets or updating modules. I'll close this for now. If I ever come back to this, I'll reopen and add detailed description.

closed 2015-07-22 by 0xc0170