ARMmbed/yotta #326: Strange network issues when depending on github repos

bogdanm, 2015-07-24:
I've noticed this recently, when working on a project that depends on things like `module:ARMmbed/module#branch`. Very often (but not always) this happens during `yt install`:

```
warning: Using head of "ns_minar_take2" branch for "atmel-rf-driver", not a tagged version
info: download atmel-rf-driver@ns_minar_take2 from GitHub ARMmbed/atmel-rf-driver-private
info: Starting new HTTPS connection (1): codeload.github.com
Traceback (most recent call last):
  File "/usr/local/bin/yt", line 4, in <module>
    yotta.main()
  File "/usr/local/lib/python2.7/dist-packages/yotta/main.py", line 213, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/dist-packages/yotta/build.py", line 41, in execCommand
    status = installAndBuild(args, following_args)
  File "/usr/local/lib/python2.7/dist-packages/yotta/build.py", line 91, in installAndBuild
    install_status = install.execCommand(args, [])
  File "/usr/local/lib/python2.7/dist-packages/yotta/install.py", line 54, in execCommand
    return installDeps(args, c)
  File "/usr/local/lib/python2.7/dist-packages/yotta/install.py", line 106, in installDeps
    test = {'own':'toplevel', 'all':True, 'none':False}[args.install_test_deps]
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 517, in satisfyDependenciesRecursive
    test = test
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 312, in __getDependenciesRecursiveWithProvider
    test = test
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 212, in __getDependenciesWithProvider
    satisfyDep, specs
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 194, in satisfyDep
    self.getName()
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 503, in provider
    r = access.satisfyVersionByInstalling(dspec.name, dspec.version_req, self.modulesPath())
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 296, in satisfyVersionByInstalling
    name, version_required, install_into, v, type=type
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 307, in _satisfyVersionByInstallingVersion
    version.unpackInto(working_directory)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/github_access.py", line 220, in unpackInto
    _getTarball(self.url, directory)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/github_access.py", line 57, in wrapped
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/github_access.py", line 129, in _getTarball
    access_common.unpackTarballStream(response, into_directory)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access_common.py", line 112, in unpackTarballStream
    for chunk in stream.iter_content(1024):
  File "/usr/lib/python2.7/dist-packages/requests/models.py", line 616, in generate
    decode_content=True):
  File "/usr/lib/python2.7/dist-packages/urllib3/response.py", line 225, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/usr/lib/python2.7/dist-packages/urllib3/response.py", line 174, in read
    data = self._fp.read(amt)
  File "/usr/lib/python2.7/httplib.py", line 573, in read
    s = self.fp.read(amt)
  File "/usr/lib/python2.7/dist-packages/urllib3/contrib/pyopenssl.py", line 152, in read
    data = self._sock.recv(left)
OpenSSL.SSL.SysCallError: (104, 'Connection reset by peer')
```

If I do `yt install` after this again, it'll work, until it finds another github/#branch dependency, then the same thing happens. Sometimes the error is a bit diffrerent:

```
warning: Using head of "ns_minar_take2" branch for "mbed-6lowpan-eventloop", not a tagged version
info: download mbed-6lowpan-eventloop@ns_minar_take2 from GitHub ARMmbed/mbed-6lowpan-eventloop
info: Starting new HTTPS connection (1): codeload.github.com
Traceback (most recent call last):
  File "/usr/local/bin/yt", line 4, in <module>
    yotta.main()
  File "/usr/local/lib/python2.7/dist-packages/yotta/main.py", line 213, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/dist-packages/yotta/build.py", line 41, in execCommand
    status = installAndBuild(args, following_args)
  File "/usr/local/lib/python2.7/dist-packages/yotta/build.py", line 91, in installAndBuild
    install_status = install.execCommand(args, [])
  File "/usr/local/lib/python2.7/dist-packages/yotta/install.py", line 54, in execCommand
    return installDeps(args, c)
  File "/usr/local/lib/python2.7/dist-packages/yotta/install.py", line 106, in installDeps
    test = {'own':'toplevel', 'all':True, 'none':False}[args.install_test_deps]
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 517, in satisfyDependenciesRecursive
    test = test
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 312, in __getDependenciesRecursiveWithProvider
    test = test
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 212, in __getDependenciesWithProvider
    satisfyDep, specs
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 194, in satisfyDep
    self.getName()
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 503, in provider
    r = access.satisfyVersionByInstalling(dspec.name, dspec.version_req, self.modulesPath())
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 296, in satisfyVersionByInstalling
    name, version_required, install_into, v, type=type
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 307, in _satisfyVersionByInstallingVersion
    version.unpackInto(working_directory)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/github_access.py", line 220, in unpackInto
    _getTarball(self.url, directory)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/github_access.py", line 57, in wrapped
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/github_access.py", line 129, in _getTarball
    access_common.unpackTarballStream(response, into_directory)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access_common.py", line 112, in unpackTarballStream
    for chunk in stream.iter_content(1024):
  File "/usr/lib/python2.7/dist-packages/requests/models.py", line 616, in generate
    decode_content=True):
  File "/usr/lib/python2.7/dist-packages/urllib3/response.py", line 225, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/usr/lib/python2.7/dist-packages/urllib3/response.py", line 174, in read
    data = self._fp.read(amt)
  File "/usr/lib/python2.7/httplib.py", line 573, in read
    s = self.fp.read(amt)
  File "/usr/lib/python2.7/dist-packages/urllib3/contrib/pyopenssl.py", line 152, in read
    data = self._sock.recv(left)
OpenSSL.SSL.ZeroReturnError
```

Again, if I do `yt install` again, it works.
I don't believe this is a network issue; it tends to happen only for github-based dependencies like the ones above, on an otherwise stable connection.
Issue appears while using yotta 0.5.1 under Linux Mint 17.1.

autopulated, 2015-07-24:
That's interesting — What Python version is it?

bogdanm, 2015-07-24:
Python 2.7.6
And yes, it's quite interesting :)

autopulated, 2015-07-24:
Can you reproduce this if you upgrade to >= 2.7.9?

autopulated, 2015-07-29:
closing unless this can be reproduced on an up-to-date patch release of python

closed 2015-07-29 by autopulated