ARMmbed/yotta #341: Can't install module globally in Windows 7

jagomo, 2015-07-31:
when trying to install a module globally like this:

*yotta -v -v -v  install --global blinkm*

The result is:

level 2:components: Component: C:\Workspaces\Eclipse\testyotta2 installed_linked=False
debug:root: install component blinkm to C:\Python27\Lib\yotta_modules
debug:access: check path C:\Python27\Lib\yotta_modules\blinkm for blinkm
level 2:components: Component: C:\Python27\Lib\yotta_modules\blinkm installed_linked=False
debug:access: not found blinkm locally
info: get versions for blinkm
debug:access: satisfy blinkm from modules registry
debug:access: fingerprint: 87:ed:cc:19:39:4b:7d:41:00:25:a9:b4:b5:a2:a1:f0
debug:access: token fields: {'iss': 'yotta', 'prn': '87:ed:cc:19:39:4b:7d:41:00:25:a9:b4:b5:a2:a1:f0', 'aud': 'https://registry.yottabuild.org', 'exp': '1438337738'}
debug:access: encoded token: eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJ5b3R0YSIsInBybiI6Ijg3OmVkOmNjOjE5OjM5OjRiOjdkOjQxOjAwOjI1OmE5OmI0OmI1OmEyOmExOmYwIiwiYXVkIjoiaHR0cHM6Ly9yZWdpc3RyeS55b3R0YWJ1aWxkLm9yZyIsImV4cCI6IjE0MzgzMzc3MzgifQ.IRM4CU8utwAyB0Kvl_UeDArq-_cUmIyJvz4PgC6vG_N5cUhptocc548uwb_DBRjh54VFXlQz8E2tc0dm0LDJkl28HqY8-0AxyXK7Ezt5_sbCr1WH-CLlK9tpPFriWNadEacmNEOQfwKJKNHrIOwyLNEOHqF-T3a6WuVgcFzmwFG7NHCwQhzoHbGBpLqYEGxRr4UPtgT2sHH_f9c3k6oEhUime2cNfVFfGGRPdHFQxKAxBYeg764TWtpUe_yY4848edR9mjeFrr1JwpYeUxKmUCL6aH4eMxWBchjVf9lZAnVzz_0Iy333coZ64SxRVq8_2azma_hezY0asDFAreUmZQ
debug:access: GET https://registry.yottabuild.org/modules/blinkm/versions, {'Authorization': u'Bearer eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJ5b3R0YSIsInBybiI6Ijg3OmVkOmNjOjE5OjM5OjRiOjdkOjQxOjAwOjI1OmE5OmI0OmI1OmEyOmExOmYwIiwiYXVkIjoiaHR0cHM6Ly9yZWdpc3RyeS55b3R0YWJ1aWxkLm9yZyIsImV4cCI6IjE0MzgzMzc3MzgifQ.IRM4CU8utwAyB0Kvl_UeDArq-_cUmIyJvz4PgC6vG_N5cUhptocc548uwb_DBRjh54VFXlQz8E2tc0dm0LDJkl28HqY8-0AxyXK7Ezt5_sbCr1WH-CLlK9tpPFriWNadEacmNEOQfwKJKNHrIOwyLNEOHqF-T3a6WuVgcFzmwFG7NHCwQhzoHbGBpLqYEGxRr4UPtgT2sHH_f9c3k6oEhUime2cNfVFfGGRPdHFQxKAxBYeg764TWtpUe_yY4848edR9mjeFrr1JwpYeUxKmUCL6aH4eMxWBchjVf9lZAnVzz_0Iy333coZ64SxRVq8_2azma_hezY0asDFAreUmZQ'}
debug:access: RegistryThingVersion modules/blinkm data: OrderedDict([(u'dependencies', OrderedDict([(u'mbed', u'ARMmbed/mbed-sdk')])), (u'licenses', [OrderedDict([(u'type', u'Apache-2.0'), (u'url', u'https://spdx.org/licenses/Apache-2.0')])]), (u'homepage', u'https://github.com/stephenpaulger/blinkm'), (u'repository', OrderedDict([(u'type', u'git'), (u'url', u'git@github.com:stephenpaulger/blinkm.git')])), (u'author', u'Stephen Paulger'), (u'keywords', [u'blinkm', u'LED', u'i2c']), (u'description', u'Control the BlinkM LED Module over i2c'), (u'version', u'0.0.1'), (u'name', u'blinkm'), (u'readme_url', u'https://yotta.blob.core.windows.net/modules/blinkm-0.0.1-readme.md'), (u'hash', OrderedDict([(u'sha256', u'677370155dcfd9e6624d144a209a2a7184d41ae03ffbf4bbb70e2570bf59eca0')])), (u'created', u'2015-01-28T19:31:28.984Z'), (u'type', u'module')])
debug:access: * selected blinkm@0.0.1 from https://registry.yottabuild.org from [blinkm@0.0.1 from https://registry.yottabuild.org]
info: download blinkm@0.0.1 from https://registry.yottabuild.org
debug:access: registry: get: https://registry.yottabuild.org/modules/blinkm/versions/0.0.1/tarball
debug:access: fingerprint: 87:ed:cc:19:39:4b:7d:41:00:25:a9:b4:b5:a2:a1:f0
debug:access: token fields: {'iss': 'yotta', 'prn': '87:ed:cc:19:39:4b:7d:41:00:25:a9:b4:b5:a2:a1:f0', 'aud': 'https://registry.yottabuild.org', 'exp': '1438337740'}
debug:access: encoded token: eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJ5b3R0YSIsInBybiI6Ijg3OmVkOmNjOjE5OjM5OjRiOjdkOjQxOjAwOjI1OmE5OmI0OmI1OmEyOmExOmYwIiwiYXVkIjoiaHR0cHM6Ly9yZWdpc3RyeS55b3R0YWJ1aWxkLm9yZyIsImV4cCI6IjE0MzgzMzc3NDAifQ.FFVaLLllcjfjHTAljsGgeze3TvwsYLnJzInIDYJ8TBEuOq78EvUP4sbGBe1OYYbJ2E0KSX9eeb-C3noUCdcQjZMc4V_VkPg9Rlmy3ML4_YtHZz5kpkKZVLWQQRudexa7Sq3SK6BRS5mazsnBn5BTIK1IFyZKyWMphc2VRpHQU-0xf6ypyZYZ4CF8w3_mrJCw0pb5SJV-R-4AzfbivKJV31FEvFlxXfFUO6uEyJo_qHqW91HamT_Bdz0M9x3GTLvPtHa3_5MogS4blND3TF-mHj5H-IzSbH9J2goa9qFqtx3zn26SgLME9fs9AJt0UiAwB1T-0LSQyoQsz9EOPshehg
debug:access: GET https://registry.yottabuild.org/modules/blinkm/versions/0.0.1/tarball, {'Authorization': u'Bearer eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJ5b3R0YSIsInBybiI6Ijg3OmVkOmNjOjE5OjM5OjRiOjdkOjQxOjAwOjI1OmE5OmI0OmI1OmEyOmExOmYwIiwiYXVkIjoiaHR0cHM6Ly9yZWdpc3RyeS55b3R0YWJ1aWxkLm9yZyIsImV4cCI6IjE0MzgzMzc3NDAifQ.FFVaLLllcjfjHTAljsGgeze3TvwsYLnJzInIDYJ8TBEuOq78EvUP4sbGBe1OYYbJ2E0KSX9eeb-C3noUCdcQjZMc4V_VkPg9Rlmy3ML4_YtHZz5kpkKZVLWQQRudexa7Sq3SK6BRS5mazsnBn5BTIK1IFyZKyWMphc2VRpHQU-0xf6ypyZYZ4CF8w3_mrJCw0pb5SJV-R-4AzfbivKJV31FEvFlxXfFUO6uEyJo_qHqW91HamT_Bdz0M9x3GTLvPtHa3_5MogS4blND3TF-mHj5H-IzSbH9J2goa9qFqtx3zn26SgLME9fs9AJt0UiAwB1T-0LSQyoQsz9EOPshehg'}
debug:access: calculated sha256 hash: 677370155dcfd9e6624d144a209a2a7184d41ae03ffbf4bbb70e2570bf59eca0 check against: 677370155dcfd9e6624d144a209a2a7184d41ae03ffbf4bbb70e2570bf59eca0
debug:access: wrote tarfile of size: 1108 to C:\Python27\Lib\yotta_modules\tmpadev67\download.tar.gz
debug:access: got file, extract into C:\Python27\Lib\yotta_modules\tmpadev67 (for C:\Python27\Lib\yotta_modules\blinkm)
debug:access: extraction complete C:\Python27\Lib\yotta_modules\blinkm
level 2:components: Component: C:\Python27\Lib\yotta_modules\blinkm installed_linked=False
Traceback (most recent call last):
  File "C:\Python27\Scripts\yotta-script.py", line 9, in <module>
    load_entry_point('yotta==0.4.5', 'console_scripts', 'yotta')()
  File "C:\Python27\lib\site-packages\yotta\main.py", line 189, in main
    status = args.command(args, following_args)
  File "C:\Python27\lib\site-packages\yotta\install.py", line 62, in execCommand
    return installComponent(args)
  File "C:\Python27\lib\site-packages\yotta\install.py", line 208, in installComponent
    os.chdir(component_name)
WindowsError: [Error 2] The system cannot find the file specified: 'blinkm'

The equivalent line of this error in the current install.py file is 196:
https://github.com/ARMmbed/yotta/blob/8abfb450aec71e80ebd4cbc2eed1efca96a7e004/yotta/install.py#L196

autopulated, 2015-07-31:
Thanks for reporting.

A module that has been installed globally cannot be used without using `yotta link` to link it into a particular application. So the right behaviour when running this command in an existing project is to automatically create that link after installing.

Please avoid using --global for now.

jagomo, 2015-07-31:
You're welcome :)

Just to clarify (and maybe help). I tried to execute it outside a project folder for a reason. If I execute the same command in an existing project, a "yotta_modules" folder is created inside the project folder and the module is installed there. Looks like the --global gets ignored in that case since no "C:\Python27\Lib\yotta_modules" folder gets created at all.

autopulated, 2015-07-31:
ah OK, got it.

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-899