ARMmbed/yotta #342: add contributing.md with link to CLA

autopulated, 2015-07-31:
Github shows a banner asking people who open pull reqs to read this, so might as well use that.

 * [CLA](https://developer.mbed.org/contributor_agreement/)
 * [github docs on contributing.md](https://help.github.com/articles/setting-guidelines-for-repository-contributors/)

hugovincent, 2015-07-31:
You mean CONTRIBUTING.md (case matters apparently):
> In the file name field, type CONTRIBUTING or CONTRIBUTING.md (with all caps!)
(https://help.github.com/articles/setting-guidelines-for-repository-contributors/)

bogdanm, 2015-07-31:
Good idea. I think we should add that to all repos in ARMmbed.

bogdanm, 2015-07-31:
(all public ones, that is)

autopulated, 2015-07-31:
:cry: all caps

bogdanm, 2015-07-31:
:sob: OH WELL, IT'LL HAVE TO DO.

hugovincent, 2015-07-31:
Call me oldschool, but for files like this I prefer all-caps... helps visually distinguish the boilerplate in all repos (LICENSE, CONTRIBUTING, README) from the content.

autopulated, 2015-07-31:
Also worth mentioning [clahub](https://www.clahub.com), which adds a commit status for whether people have agreed to the CLA – we use it for https://github.com/coggle/coggle-translations, has worked really well so far.

PR #416 (2015-09-17): Add CONTRIBUTING.md
Resolves #342
commit c3607c65 (2015-09-17): Add CONTRIBUTING.md

Resolves #342
closed 2015-09-17 by autopulated