ARMmbed/yotta #349: yt test - does not close the port (windows)

0xc0170, 2015-08-03:
Using frdm-k64f, windows 7, yotta v 0.5.0, mbedgt 0.0.10

The first test runs, all the tests after the first one fails to obtain com port. If I run mbedgt directly, it works.

```
 yt test
info: generate for target: frdm-k64f-gcc 0.0.21 at xxxx\yotta_targets\frdm-k64f-gcc
GCC version is: 4.8.4
-- Configuring done
-- Generating done
-- Build files have been written to: xxxxx/build/frdm-k64f-gcc
ninja: no work to do.
info: using filter "mbedgt --digest stdin -v -V" for tests in xxxxx\build\frdm-k64f-gcc\test info: test mbed-core: xxxxxx
MBED: Instrumentation: "COM44" and disk: "E:"
HOST: Copy image onto target...
        1 file(s) copied.
HOST: Initialize serial port...
...port ready!
HOST: Reset target...
HOST: Unknown property: ========== Starting event handler test ==========

// some more text

Error: test mbed-core-test-eventhandler failed
info: test mbed-core: xxxxxxx
MBED: Instrumentation: "COM44" and disk: "E:"
HOST: Copy image onto target...
        1 file(s) copied.
HOST: Initialize serial port...
........................................MBED: could not open port 'COM44': WindowsError(5, 'Access is denied.')

{{ioerr_serial}}
{{end}}
Hmbed-host-test-runner: Stopped
error: test mbed-core-test-functionpointer failed
```

autopulated, 2015-08-03:
Please try this: https://github.com/ARMmbed/yotta/pull/350

With `yotta -vvv test` it should provide useful output about what's happening, including whether yotta is killing the test child, or it is exiting gracefully.

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-898