ARMmbed/yotta #353: yt update - error directory is not empty [windows error]

0xc0170, 2015-08-04:
yt version (0.5.0), windows 7.

Running yotta update from time to time ends up in the traceback with an error "The directory is not empty". The snippet below shows it:

```
yt update
info: get versions for frdm-k64f-gcc
info: update outdated: frdm-k64f-gcc@0.0.16 -> frdm-k64f-gcc@0.0.21 from https://registry.yottabuild.org
info: download frdm-k64f-gcc@0.0.21 from https://registry.yottabuild.org
info: get versions for mbed-gcc
info: download mbed-gcc@0.0.8 from https://registry.yottabuild.org
info: get versions for xxxxx
// text: get more modules 
info: update outdated: mbed-hal-ksdk-mcu@0.1.4 -> mbed-hal-ksdk-mcu@0.1.5 from https://yotta-private.herokuapp.com
Traceback (most recent call last):
  File "C:\Python27\Scripts\yt-script.py", line 9, in <module>
    load_entry_point('yotta==0.5.0', 'console_scripts', 'yt')()
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\main.py", line 213, in main
    status = args.command(args, following_args)
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\update.py", line 28, in execCommand
    updateDeps(args)
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\update.py", line 51, in updateDeps
    available_components = [(c.getName(), c)]
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\component.py", line 517, in satisfyDependenciesRecursive
    test = test
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\component.py", line 333, in __getDependenciesRecursiveWithProvider
    _processed = _processed
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\component.py", line 333, in __getDependenciesRecursiveWithProvider
    _processed = _processed
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\component.py", line 333, in __getDependenciesRecursiveWithProvider
    _processed = _processed
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\component.py", line 333, in __getDependenciesRecursiveWithProvider
    _processed = _processed
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\component.py", line 312, in __getDependenciesRecursiveWithProvider
    test = test
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\component.py", line 212, in __getDependenciesWithProvider
    satisfyDep, specs
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\component.py", line 194, in satisfyDep
    self.getName()
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\component.py", line 481, in provider
    r = access.satisfyVersionFromSearchPaths(dspec.name, dspec.version_req, search_dirs, update_if_installed)
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\access.py", line 285, in satisfyVersionFromSearchPaths
    fsutils.rmRf(local_version.path)
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\fsutils.py", line 45, in rmRf
    shutil.rmtree(path, onerror=fixPermissions)
  File "C:\Python27\lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Python27\lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Python27\lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Python27\lib\shutil.py", line 256, in rmtree
    onerror(os.rmdir, path, sys.exc_info())
  File "C:\Python27\lib\shutil.py", line 254, in rmtree
    os.rmdir(path)
WindowsError: [Error 145] The directory is not empty: u'xxxx\\yotta_modules\\mbed-hal-ksdk-mcu\\mbed-hal-ksdk-mcu\\TARGET_KSDK_CODE\\drivers'
```

The second immediate yt update fixes this:
```
yt update
info: get versions for frdm-k64f-gcc
// text more modules
info: update outdated: mbed-hal-ksdk-mcu@0.1.4 -> mbed-hal-ksdk-mcu@0.1.5 from https://yotta-private.herokuapp.com
info: download mbed-hal-ksdk-mcu@0.1.5 from https://yotta-private.herokuapp.com
```

autopulated, 2015-08-04:
I think this was probably caused by a file being open from the directory (or the directory being open in explorer)

Not sure there's anything we should do about that causing a failure, but the backtrace can definitely be converted into a nicer error.

0xc0170, 2015-08-04:
I can't confirm that I had any file from that repository opened anywhere. I will do once I hit this back again or someone else might be faster.

Why it does not fail to update the second time?

autopulated, 2015-08-04:
That I'm not sure: does seem to suggest something else is going on.

0xc0170, 2015-08-05:
I just got this error again. I removed yotta_modules, build (should get new ones):

```
info: download xxxx-platform@0.3.0 from https://yotta-private.herokuapp.com
Traceback (most recent call last):
  File "C:\Python27\Scripts\yt-script.py", line 9, in <module>
    load_entry_point('yotta==0.5.0', 'console_scripts', 'yt')()
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\main.py", line 213, in main
    status = args.command(args, following_args)
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\build.py", line 41, in execCommand
    status = installAndBuild(args, following_args)
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\build.py", line 91, in installAndBuild
    install_status = install.execCommand(args, [])
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\install.py", line 54, in execCommand
    return installDeps(args, c)
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\install.py", line 106, in installDeps
    test = {'own':'toplevel', 'all':True, 'none':False}[args.install_test_deps]
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\component.py", line 517, in satisfyDependenciesRecursive
    test = test
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\component.py", line 333, in __getDependenciesRecursiveWithProvider
    _processed = _processed
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\component.py", line 333, in __getDependenciesRecursiveWithProvider
    _processed = _processed
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\component.py", line 312, in __getDependenciesRecursiveWithProvider
    test = test
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\component.py", line 212, in __getDependenciesWithProvider
    satisfyDep, specs
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\component.py", line 194, in satisfyDep
    self.getName()
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\component.py", line 503, in provider
    r = access.satisfyVersionByInstalling(dspec.name, dspec.version_req, self.modulesPath())
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\access.py", line 296, in satisfyVersionByInstalling
    name, version_required, install_into, v, type=type
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\access.py", line 307, in _satisfyVersionByInstallingVersion
    version.unpackInto(working_directory)
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\registry_access.py", line 342, in unpackInto
    _getTarball(self.url, directory, self.sha256)
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\registry_access.py", line 218, in _getTarball
    return access_common.unpackTarballStream(response, directory, ('sha256', sha256))
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\access_common.py", line 158, in unpackTarballStream
    fsutils.rmRf(temp_directory)
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\fsutils.py", line 45, in rmRf
    shutil.rmtree(path, onerror=fixPermissions)
  File "C:\Python27\lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Python27\lib\shutil.py", line 252, in rmtree
    onerror(os.remove, fullname, sys.exc_info())
  File "c:\code\git_repo\github\yotta_thing\yotta\yotta\lib\fsutils.py", line 37, in fixPermissions
    os.chmod(path, stat.S_IWUSR)
WindowsError: [Error 5] Access is denied: u'C:\\Code\\git_repo\\github\\yotta_thing\\mbed-example-asynch-spi-private\\yotta_modules\\tmpjlmb4b\\xxxx-platform-0.3.0\\xxx-platform'
```

The module with this error (xxx-platform) was not existent prior the build (could not be open by any tool besides what is yotta doing).

autopulated, 2015-08-05:
Ok, thanks for the update.

PR #358 (2015-09-07): possible fix for directories failing to be removed on windows (#353)
So this is a fun one. According to People On The Internet (and enough of them that this speculation seems reliable), files accessed by processes can be held open on windows by various other processes, including antivirus software, and search indexing even after they are closed by the process.

So you might need to try several times to remove files...

Sounds plausible anyway. This patch tries to workaround such problems.

Should fix #353
closed 2015-09-07 by autopulated