ARMmbed/yotta #356: Changes in header files (only) do not trigger new build

AlessandroA, 2015-08-06:
If only header files are changed, yotta does not build the executable again. This was observed both on Linux (make) and Windows (ninja). Tried with 2 different targets (`frdm-k64f-gcc` and `frdm-k64f-armcc`)

0xc0170, 2015-08-06:
I can confirm, I have been using yt clean && yt build just in case.

autopulated, 2015-08-06:
a minimal example that reproduces this would be really useful, if you have one?

autopulated, 2015-08-14:
For reference, a similar issue was previously resolved here: https://github.com/ARMmbed/target-mbed-gcc/pull/1

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-897

autopulated, 2015-11-03:
I believe this has been fixed in the target descriptions. Please re-open if it is still a problem.

closed 2015-11-03 by autopulated