ARMmbed/yotta #369: Fetching module from mercurial repo is not possible

jagomo, 2015-08-11:
According to documentation:

_Depending on hg Modules_
_To specify a module available from a mercurial server as a dependency, use a hg URL:_

_"usefulmodule": "hg+ssh://somwhere.com/anything/anywhere"_
_"usefulmodule": "hg+ssh://somwhere.com/anything/anywhere#<version specification>"_
_"usefulmodule": "<anything>://somwhere.hg"_
_"usefulmodule": "<anything>://somwhere.hg#<version specification>"_

Mi clone adress is http://someip:someport/application

If i set the dependency in my module.json as

"targetDependencies": {
      "st-nucleo-f091rc-gcc": {
           "mbed-dummy-app": "http://someip:someport/application"
      }

I get the following error message:
_ValueError: Invalid version source url: "http://someip:someport/application"_

If I change the dependency to

"targetDependencies": {
      "st-nucleo-f091rc-gcc": {
           "mbed-dummy-app": "http://someip:someport/application**.hg**"
      }

Like the documentation suggest

I get the error
_Err: abort: HTTP Error 404: Not Found_

As it's obvious since the url doesn't exists.

0xc0170, 2015-08-11:
Please for code snippets use https://help.github.com/articles/github-flavored-markdown/#fenced-code-blocks. 

Note: I am not able to edit the post to be more readable :(

jagomo, 2015-08-11:
Sorry, done. Hope is easier to understand now :)

autopulated, 2015-08-11:
Try hg+http://?

Unfortunately yotta needs this indication in the URL so that it knows to fetch over hg rather than anything else.

jagomo, 2015-08-11:
Thanks! that worked and now tries to grab it :)

Now I face a different problem. AFAIK grabbing from the tip is neither implemented for git or hg. Nevertheless I managed to grab modules from git using the syntax "#branch_name" at the end in of the URL.

From the documentation, this syntax for hg is supposed to be "#version" where I assume that version is supposed to be tag in the hg repo from the following hg_access.py code:
```
def availableVersions(self):
        r = []
        for t in self.vcs.tags():
            logger.debug("available version tag: %s", t)
            # ignore empty tags:
            if not len(t.strip()):
                continue
            try:
                r.append(HGCloneVersion(t, self))
            except ValueError:
                logger.debug('invalid version tag: %s', t)
        return r
```

So I tagged my default branch as 2.0 and now my module.json looks like this:

```
"targetDependencies": {
      "st-nucleo-f091rc-gcc": {
           "mbed-dummy-app": "hg+http://someip:someport/application#2.0"
      }
},
```

But I still get the following warning:

_warning: hg repository "hg+http://someip:someport/application#2.0" has no tagged versions, default branch will be used_

And the following error:

_Traceback (most recent call last):
  File "C:\Python27\Scripts\yotta-script.py", line 9, in <module>
    load_entry_point('yotta==0.4.5', 'console_scripts', 'yotta')()
  File "C:\Python27\lib\site-packages\yotta\main.py", line 189, in main
    status = args.command(args, following_args)
  File "C:\Python27\lib\site-packages\yotta\build.py", line 80, in execCommand
    errcode = install.execCommand(args, [])
  File "C:\Python27\lib\site-packages\yotta\install.py", line 58, in execCommand
    return installDeps(args, c)
  File "C:\Python27\lib\site-packages\yotta\install.py", line 108, in installDeps
    test = {'own':'toplevel', 'all':True, 'none':False}[args.install_test_deps]
  File "C:\Python27\lib\site-packages\yotta\lib\component.py", line 493, in satisfyDependenciesRecursive
    test = test
  File "C:\Python27\lib\site-packages\yotta\lib\component.py", line 312, in __getDependenciesRecursiveWithProvider
    test = test
  File "C:\Python27\lib\site-packages\yotta\lib\component.py", line 212, in __getDependenciesWithProvider
    satisfyDep, specs
  File "C:\Python27\lib\site-packages\yotta\lib\component.py", line 194, in satisfyDep
    self.getName()
  File "C:\Python27\lib\site-packages\yotta\lib\component.py", line 479, in provider
    r = access.satisfyVersionByInstalling(dspec.name, dspec.version_req, self.modulesPath())
  File "C:\Python27\lib\site-packages\yotta\lib\access.py", line 289, in satisfyVersionByInstalling
    v = latestSuitableVersion(name, version_required, _registryNamespaceForType(type))
  File "C:\Python27\lib\site-packages\yotta\lib\access.py", line 171, in latestSuitableVersion
    vers = [local_clone.tipVersion()]
  File "C:\Python27\lib\site-packages\yotta\lib\hg_access.py", line 67, in tipVersion
    raise NotImplementedError
NotImplementedError_

Since, yotta tries to grab the tip and the *NotImplementedError* gets raised.

autopulated, 2015-08-11:
I don't think hg supports branch access (including access to the tip of the default branch), but it should be able to fetch specific tagged versions, if you use an existing tag as your #version.

Note that "2.0" is not not a valid [semantic version](https://semver.org).

jagomo, 2015-08-11:
Thanks for the pointer, that was it :)

After using a tag with a valid semantic version it worked. For completion here is how my dependency looks now:
```
"targetDependencies": {
      "st-nucleo-f091rc-gcc": {
    	  "mbed-dummy-app": "hg+http://someip:someport/application#2.0.1"
      }
  },
```
Would be great if this clarifications could be in the yotta documentation :) (Is there any way to collaborate?)

autopulated, 2015-08-11:
Sure – you can open a pull request against the module.json reference by forking and editing this file (make a pull request against the master branch): https://github.com/ARMmbed/yotta/blob/master/docs/reference/module.md :smiley: 

Now that I look at that section of the documentation, it does say that tags and branches are not listed as supported for hg, only version specifications, but it should be possible to make that clearer.

I'm closing this as a resolved question.

closed 2015-08-11 by autopulated