ARMmbed/yotta #370: yt update does not check testTargetDependencies

mpg, 2015-08-11:
According to its output on my module, yt update only checks new versions for modules in  targetDependencies and their dependencies, but not for modules listed in testTargetDependencies. (I don't have any target-independent testDependencies, so I don't know if they are checked by yt update or not.)

If there is a good reason to exclude test (target) dependencies by default, then a new option, eg yt update --tests, should be added to allow updating them.

autopulated, 2015-08-11:
Nope, this is a bug!

mpg, 2015-10-05:
Hey, can you up the priority of this? I just got a situation where I had to `rm -r yotta_modules` before I could build successfully, because otherwise I was stuck with old conflicting versions that `yt update` wouldn't fix. Thanks!

autopulated, 2015-10-05:
If you're happy to open a PR and verify that this it works, I think this is a one line change. Try adding `test = True,` here: 
https://github.com/ARMmbed/yotta/blob/master/yotta/update.py#L49

closed 2015-10-05 by autopulated

mpg, 2015-10-05:
Thanks! I'll check that soon (no pending update now).

commit 1ffb6494 (2015-10-05): Merge pull request #458 from autopulated/master

add --test-dependencies switch to update subcommand; fixes #370
commit a029c644 (2015-10-05): add --test-dependencies switch to update subcommand; fixes #370